The-OpenROAD-Project/OpenLane #167: TritonRoute failure results in report_generation_wrapper.py endless loop

antonblanchard, 2021-01-16:
In the process of testing `cugr`, I managed to get `TritonRoute` to fail. We end up in an endless loop executing `report_generation_wrapper.py`:

```
start track assignment
Done with 136029 vertical wires in 5 frboxes and Error: assignIroute_bestTrack select no track for i_in[11] @(361.5, 394.5) (370.5, 394.5 met5 idx1/2=28/27
[ERROR]: during executing: "TritonRoute /project/openlane/icache/runs/icache/tmp/routing/23-tritonRoute.param |& tee >&@stdout /project/openlane/icache/runs/icache/logs/routing/23-tritonRoute.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

[ERROR]: Please check TritonRoute  log file
[ERROR]: Dumping to /project/openlane/icache/runs/icache/error.log
[INFO]: Generating Final Summary Report...
[ERROR]: during executing: "python3 /openLANE_flow/report_generation_wrapper.py -d /project/openlane/icache -dn icache -t icache -o /project/openlane/icache/runs/icache/reports/final_summary_report.csv -m /project/openlane/icache/runs/icache/reports/manufacturability_report.rpt -r /project/openlane/icache/runs/icache"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
sed: can't read /project/openlane/icache/runs/icache/reports/runtime.txt: No such file or directory
awk: fatal: cannot open file `/project/openlane/icache/runs/icache/results/routing/icache.def' for reading (No such file or directory)
awk: fatal: cannot open file `/project/openlane/icache/runs/icache/results/routing/icache.def' for reading (No such file or directory)
awk: fatal: cannot open file `/project/openlane/icache/runs/icache/results/routing/icache.def' for reading (No such file or directory)
awk: fatal: cannot open file `/project/openlane/icache/runs/icache/results/routing/icache.def' for reading (No such file or directory)
/openLANE_flow/scripts/report/report.sh: line 62: ((-)/1000)*((-)/1000) : syntax error: operand expected (error token is ")/1000)*((-)/1000) ")

[ERROR]: Please check python3  log file
[ERROR]: Dumping to /project/openlane/icache/runs/icache/error.log
[INFO]: Generating Final Summary Report...
[ERROR]: during executing: "python3 /openLANE_flow/report_generation_wrapper.py -d /project/openlane/icache -dn icache -t icache -o /project/openlane/icache/runs/icache/reports/final_summary_report.csv -m /project/openlane/icache/runs/icache/reports/manufacturability_report.rpt -r /project/openlane/icache/runs/icache"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
sed: can't read /project/openlane/icache/runs/icache/reports/runtime.txt: No such file or directory
awk: fatal: cannot open file `/project/openlane/icache/runs/icache/results/routing/icache.def' for reading (No such file or directory)
awk: fatal: cannot open file `/project/openlane/icache/runs/icache/results/routing/icache.def' for reading (No such file or directory)
awk: fatal: cannot open file `/project/openlane/icache/runs/icache/results/routing/icache.def' for reading (No such file or directory)
awk: fatal: cannot open file `/project/openlane/icache/runs/icache/results/routing/icache.def' for reading (No such file or directory)
```

antonblanchard, 2021-01-18:
This fixes the immediate issue for me: https://github.com/efabless/openlane/pull/168

ax3ghazy, 2021-01-18:
The main issue here is that TritonRoute failed to use the guide from `cugr` to route the design.

```
start track assignment
Done with 136029 vertical wires in 5 frboxes and Error: assignIroute_bestTrack select no track for i_in[11] @(361.5, 394.5) (370.5, 394.5 met5 idx1/2=28/27
[ERROR]: during executing: "TritonRoute /project/openlane/icache/runs/icache/tmp/routing/23-tritonRoute.param |& tee >&@stdout /project/openlane/icache/runs/icache/logs/routing/23-tritonRoute.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

Could you share the DEF file (still under `runs/icache/tmp/routing/fastroute.def`) and the guides file produced (`runs/icache/tmp/routing/fastroute.guide`). I will be changing file names to be more tool-agnostic in the next couple of commits.

agorararmard, 2021-01-18:
@antonblanchard: The other two parts of the issue are mentioned in #168 as follows:
I implemented a fix in 0f936f9 for the first part of the issue (the recursive calling), and another fix for the other half of the issue here: 8beb141 which allows for better support for generating reports for incomplete runs.

However, I won't close this issue because it's flagging another issue with cugr as @ax3ghazy mentioned. So, we'll keep it open until you share the DEF file and the GUIDES as @ax3ghazy requested and this is resolved.

antonblanchard, 2021-01-19:
I'm trying to reproduce, but getting an earlier failure in cugr:

```
terminate called after throwing an instance of 'Rsyn::Exception'
  what():  Only in and out direction are supported right now.
```

[multiply_4.tar.gz](https://github.com/efabless/openlane/files/5838080/multiply_4.tar.gz)
[merged_unpadded.lef.gz](https://github.com/efabless/openlane/files/5838089/merged_unpadded.lef.gz)
[17-addspacers.obs.def.gz](https://github.com/efabless/openlane/files/5838090/17-addspacers.obs.def.gz)

agorararmard, 2021-01-22:
@antonblanchard: I think it would be nice to open another issue about CU-GR and close this one for traceability (issue title).

closed 2021-01-24 by antonblanchard