The-OpenROAD-Project/OpenLane #173: yosys synthesis thinks macro is combinatorial

antonblanchard, 2021-01-22:
Looking through the log files, I notice `yosys` complains that my macro is combinatorial (it isn't):

```
ABC: + retime -D -D 15000 -M 6 
ABC: + scleanup 
ABC: Error: The network is combinational.
```

We call retime again later on (with a different strategy), and this time it works:

```
ABC: + fraig_store 
ABC: + fraig_restore 
ABC: + map 
ABC: + 
ABC: + retime -D -D 15000 
```

Are we missing a step before the first retime that inserts sequential elements?

Ravenslofty, 2021-02-17:
Look at your logs again: it's not `retime` complaining, it's `scleanup`.

An `Error: the network is combinational` from ABC generally means that the Yosys `abc` pass hasn't been given the `-dff` argument, and thus instead of passing (some) flops into the netlist, Yosys turns the inputs and outputs of the flops into primary outputs and primary inputs of the netlist. Thus the netlist is combinational and there is nothing for a sequential pass to do.

From reading the source code, it seems like `retime` on a combinational network is a silent no-op, but perhaps it should complain more obviously.

antonblanchard, 2021-02-17:
Yep, I've been experimenting with the openroad flow after adding `-dff` and as expected it's giving a big improvement in timing for my pipelined multiplier test. We might want to add an openlane option to add `-dff` and do register retiming on macros

Ravenslofty, 2021-02-17:
Just keep in mind that ABC can sometimes be pathologically slow when operating on flops; `scorr` is notorious for this in my experience.

antonblanchard, 2021-02-17:
Good to know. We could add a warning to the option so people don't flip it randomly.

closed 2021-09-01 by donn