The-OpenROAD-Project/OpenLane #174: Quick Start instructions fail on `make pdk` with "the input device is not a TTY"

tommythorn, 2021-01-22:
Looks like a Docker related issue.

$ docker --version
Docker version 19.03.13, build 4484c46

Full build:
[build.log](https://github.com/efabless/openlane/files/5854062/build.log)
](url)

antonblanchard, 2021-01-22:
Removing `-ti` from the `docker run` command would likely fix this.

tommythorn, 2021-01-22:
Thanks, I reran that command without the `-it` arg and it appeared to complete successfully, but the subsequent `make test` was a little unhappy:
```
  make test
cd /u/openlane && \
        docker run -it -v /u/openlane:/openLANE_flow -v /u:/u -e PDK_ROOT=/u -u 502:502 openlane:rc6 sh -c "./flow.tcl -design spm -tag openlane_test -disable_output -overwrite"
[INFO]: 
        ___   ____   ___  ____   _       ____  ____     ___
        /   \ |    \ /  _]|    \ | |     /    ||    \   /  _]
        |     ||  o  )  [_ |  _  || |    |  o  ||  _  | /  [_
        |  O  ||   _/    _]|  |  || |___ |     ||  |  ||    _]
        |     ||  | |   [_ |  |  ||     ||  _  ||  |  ||   [_
        \___/ |__| |_____||__|__||_____||__|__||__|__||_____|

[INFO]: Version: mpw-one-b
[INFO]: Running non-interactively
[INFO]: Using design configuration at /openLANE_flow/designs/spm/config.tcl
[INFO]: Sourcing Configurations from /openLANE_flow/designs/spm/config.tcl
[INFO]: PDKs root directory: /u
[INFO]: PDK: sky130A
[INFO]: Setting PDKPATH to /u/sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Sourcing Configurations from /openLANE_flow/designs/spm/config.tcl
[INFO]: Current run directory is /openLANE_flow/designs/spm/runs/openlane_test
[WARNING]: Removing exisiting run /openLANE_flow/designs/spm/runs/openlane_test
[INFO]: Preparing LEF Files
[INFO]: Trimming Liberty...
[ERROR]: during executing: "/openLANE_flow/scripts/libtrim.pl /u/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib /u/sky130A/libs.tech/openlane/sky130_fd_sc_hd/no_synth.cells > /openLANE_flow/designs/spm/runs/openlane_test/tmp/trimmed.lib"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
Couldn't open /u/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib at /openLANE_flow/scripts/libtrim.pl line 37.

[ERROR]: Please check /openLANE_flow/scripts/libtrim.pl  log file
[ERROR]: Dumping to /openLANE_flow/designs/spm/runs/openlane_test/error.log

    while executing
"try_catch $::env(SCRIPTS_DIR)/libtrim.pl $arg_values(-input) $scl_no_synth_lib > $arg_values(-output)"
    (procedure "trim_lib" line 14)
    invoked from within
"trim_lib"
    (procedure "prep" line 183)
    invoked from within
"prep {*}$args"
    (procedure "run_non_interactive_mode" line 9)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
        puts_info "Running interactively"
        if { [info exists arg_values(-file)..."
    (file "./flow.tcl" line 164)
make: *** [Makefile:136: test] Error 1
```

agorararmard, 2021-01-22:
@tommythorn: from looking at the build log provided above, this is not a docker issue (at least the part showing in the log isn't), rather an issue with [skywater-pdk](https://github.com/google/skywater-pdk)'s `make timing`: 
```
ConfigurationLoadError: Unable to load configuration file.
  path: /u/skywater-pdk/env/conda/.condarc
  reason: invalid yaml at line 3, column 0
```
I have seen this previously in the CI, however never faced the issue myself locally. It just happens that the fetched file contents are not valid. The straight forward solution to do is just to delete the PDK_ROOT and re-run `make pdk` again. However, it would be very appreciated if you open an issue at [skywater-pdk](https://github.com/google/skywater-pdk) and post the contents of that file: `/u/skywater-pdk/env/conda/.condarc`.

tommythorn, 2021-01-22:
Thanks Amr,

I'd be happy to, but I'm not sure even what subject to use.  Two additional details:

1. The full original build log: https://gist.github.com/tommythorn/8bd1a2ca9cc06d89347f01fd8bcc885b
2. The contents of `/u/skywater-pdk/env/conda/.condarc`:
```
envs_dirs:
  - /u/skywater-pdk/env/conda/envs
onda-pkgs
```

tommythorn, 2021-01-25:
The issue was the presence of a `MAKEFLAGS` environment variable set to `-j24`.  This breaks openlane and skywater-pdk with isn't parallel make safe.  It's fixable but also trivial to workaround, so closing.

closed 2021-01-25 by tommythorn

mithro, 2021-01-25:
@tommythorn -- It would probably be good if we detect the usage of `-j` in the Makefile and error out?

Also I think this issue should have been on https://github.com/google/skywater-pdk right?

tommythorn, 2021-01-25:
@mithro, the original issue was exclusive to openlane.  While it might have been related to `-j`, it hasn't been root caused AFAIK.  The 2ndary issues turned out to be a skywater-pdk issue so I opened #295 there.

mithro, 2021-01-25:
The issue is now at https://github.com/google/skywater-pdk/issues/295