The-OpenROAD-Project/OpenLane #176: Diode strategy 2 (and 5) fails in spice extraction (illegal overlaps)

antonblanchard, 2021-01-24:
The following test case [icache.tar.gz](https://github.com/efabless/openlane/files/5862927/icache.tar.gz) fails with:

```
Extracting icache into icache.ext:
icache: 210688 fatal errors
Total of 210688 errors (check feedback entries).
exttospice finished.
Using technology "sky130A", version 1.0.110-0-gd6807dc
[ERROR]: There are illegal overlaps (e.g., routes over obstructions) in your design.
[ERROR]: See /openLANE_flow/designs/icache/runs/24-01_03-47/logs/magic/43-magic_ext2spice.feedback.txt for more.
```

agorararmard, 2021-01-27:
@antonblanchard: This is allegedly fixed with the latest open_pdks version from opencircuitdesign.com. Can you please confirm?

antonblanchard, 2021-01-28:
Thanks, I think this is fixed. I've tested a couple of my macros and openlane runs without any DRC and LVS issues, and I see a mix of real diodes and fake diodes in the final output. Closing.

closed 2021-01-28 by antonblanchard