The-OpenROAD-Project/OpenLane #179: TritonRoute creates new via2_FR/via4_FR vias

antonblanchard, 2021-01-28:
This doesn't appear to cause problems, but it's a bit strange that TritonRoute creates new via2_FR/via4_FR vias. Ahmed confirms that we should be providing it in the techLEF:

```
Warning: met2 does not have viaDef align with layer direction, generating new viaDef via2_FR...
Warning: met4 does not have viaDef align with layer direction, generating new viaDef via4_FR...
```

antonblanchard, 2021-03-09:
I chased this down and submitted a PR to OpenROAD: https://github.com/The-OpenROAD-Project/OpenROAD/pull/595

antonblanchard, 2021-03-23:
The fix is now upstream in OpenROAD, so we can close this once openlane does an update.

donn, 2021-09-01:
OpenROAD was since updated,  closed.

closed 2021-09-01 by donn