The-OpenROAD-Project/OpenLane #215: read_lef error

troysung, 2021-02-07:
when init_floorplan, after read_elf ,error occured 
[INFO]: Synthesis was successful
[INFO]: Running Floorplanning...
[INFO]: Running Initial Floorplanning...
[INFO]: current step index: 3
[INFO]: /openLANE_flow/scripts
OpenROAD 0.9.0 1415572a73
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Warning: /root/ic/skywater-pdk/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib line 31, default_operating_condition tt_025C_1v80 not found.
/openLANE_flow/designs/inverter/runs/07-02_11-15/tmp/merged_unpadded.lef
Notice 0: Reading LEF file:  /openLANE_flow/designs/inverter/runs/07-02_11-15/tmp/merged_unpadded.lef
Notice 0:     Created 11 technology layers
Notice 0:     Created 25 technology vias
Notice 0:     Created 437 library cells
Notice 0: Finished LEF file:  /openLANE_flow/designs/inverter/runs/07-02_11-15/tmp/merged_unpadded.lef
[ERROR]: during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_floorplan.tcl |& tee >&@stdout /openLANE_flow/designs/inverter/runs/07-02_11-15/logs/floorplan/3-verilog2def.openroad.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: segmentation violation

[ERROR]: Please check openroad  log file
[ERROR]: Dumping to /openLANE_flow/designs/inverter/runs/07-02_11-15/error.log

agorararmard, 2021-02-08:
@troysung: We're going to need more context. Openlane version, configurations, design, pdk version, and so on.

agorararmard, 2021-02-15:
@troysung: Any resolution in this?

donn, 2021-06-16:
Stale, closed.

closed 2021-06-16 by donn