The-OpenROAD-Project/OpenLane #231: Issue Regarding Placement error in OPENLANE RTL to GDS flow

DarshanGuled, 2021-02-11:
[image]
Hello,
This is Darshan here.
We are working on OPENLANE FLOW. We have installed and tested example of SPM which was already present in the design folder.
Now we are trying to add our own design like counter and D - FF design, but we are facing the following errors in the in Detailed Placement.
Please Anyone help is to debug the error``

agorararmard, 2021-02-11:
Hi @DarshanGuled: Have you checked [this](https://github.com/efabless/openlane/wiki#what-to-do-if-the-flow-failed-in-any-of-the-steps-from-floorplan-up-to-routing)? 
In a nutshell, lower your `FP_CORE_UTIL` and `PL_TARGET_DENSITY`; defaults are 50 and 0.55, respectively.
You can also try setting `CELL_PAD` to 4. However, this may not be needed, depending on the OpenLane/open_pdks version you're using.

[This](https://github.com/efabless/openlane/wiki#how-to-get-to-0-drc-violations) is also relevant.

DarshanGuled, 2021-02-11:
> Hi @DarshanGuled: Have you checked [this](https://github.com/efabless/openlane/wiki#what-to-do-if-the-flow-failed-in-any-of-the-steps-from-floorplan-up-to-routing)?
> In a nutshell, lower your [...]

Thank you for the reply @agorararmard I have set the default FP_CORE_UTIL and PL_TARGET_DENSITY Its still giving same error. When i try to change the value and try again and again. It passed the placement error but its showing  
Instance cannot fit in to the core area 
What can i do for this error??!

agorararmard, 2021-02-11:
@DarshanGuled: You should keep reducing the values until they work. An exploration script would be useful in that case. [This](https://github.com/efabless/openlane/wiki#how-to-get-to-0-drc-violations) is still relevant.

DarshanGuled, 2021-02-11:
> @DarshanGuled: You should keep reducing the values until they work. An exploration script would be useful in that case. [This](https://github.com/efabless/openlane/wiki#how-to-get-to-0-drc-violations) is still relevant.

Thank you soo much @agorararmard This really helps me alot. One last thing if we are facing the OVERLAPPING IS FAILED in Openlane placement process then what can we do to eliminate that error?.

Thank you so much for the information. The links are really helpful to understand the constraints.

agorararmard, 2021-02-11:
@DarshanGuled: Also, the general solution to that is to reduce `FP_CORE_UTIL` and `PL_TARGET_DENSITY`. It's advisable to have PL_TARGET_DENSITY=(FP_CORE_UTIL+5)/100. (i.e the default values are 50, 0.55).

Also, make sure your `CELL_PAD` is set to 4.

agorararmard, 2021-02-15:
@DarshanGuled: Did you manage to resolve the issue?

agorararmard, 2021-02-22:
@DarshanGuled: I will assume this issue is resolved and close it. Please reopen if the issue still stands.

DarshanGuled, 2021-02-23:
Hello sir,
I have attached 2 snaps 
One consisting of error which we are facing.
And another one is .sdc file of our design.
Sir till Technology mapping its working but after that we are getting the above error.
Please help us to resolve the error

agorararmard, 2021-02-23:
@DarshanGuled: I'm unable to view your second snap. Please try re-uploading it.

As for the error above, this is still the solution:

> @DarshanGuled: Also, the general solution to that is to reduce `FP_CORE_UTIL` and `PL_TARGET_DENSITY`. It's advisable to have PL_TARGET_DENSITY=(FP_CORE_UTIL+5)/100. (i.e the default values are 50, 0.55).
> 
> Also, make sure your `CELL_PAD` is set to 4.

If you could provide me with the design and your configurations I can look further into it.

On that note, which openlane version are you using?

harshithsn, 2021-06-10:
> Hi @DarshanGuled: Have you checked [this](https://github.com/efabless/openlane/wiki#what-to-do-if-the-flow-failed-in-any-of-the-steps-from-floorplan-up-to-routing)?
> In a nutshell, lower your [...]

How to change FP_CORE_UTIL and PL_TARGET_DENSITY & CELL_PAD value?
And in which file i need to change /edit it?

harshithsn, 2021-06-10:
> Hi @DarshanGuled: Have you checked [this](https://github.com/efabless/openlane/wiki#what-to-do-if-the-flow-failed-in-any-of-the-steps-from-floorplan-up-to-routing)?
> In a nutshell, lower your [...]

How to change FP_CORE_UTIL and PL_TARGET_DENSITY & CELL_PAD value?
And in which file do i need to change /edit it? @DarshanGuled @agorararmard 
please help me!

DarshanGuled, 2021-06-10:
Hello @harshithsn .
In Openlane folder there is one folder called configuration in that go to Placement and you can see all configuration file select placement.tcl file and change the values

vanishka-tanwar, 2021-08-15:
Hello OL Team,

As, I am working in openlane , and is trying to make the Ip layout design of my selected circuit but when I am executing it so, at the time of execution of my design file its showing this same error to me at the time of placement 

[image]

[image]

as I have also tried it by inserting this command 
set ::env(PL_TARGET_DENSITY) "0.68"
into config.tcl file of my design but still the same problem perist

harshithsn, 2021-08-15:
@vanishka-tanwar 
Try to change FP_CORE_UTIL value.
which design are you executing ? Design Name?

vanishka-tanwar, 2021-08-16:
@harshithsn 
How much value should I give to Fp_CORE_UTIL value
My design is 4 bit magintude comparator and I saved it by this name dvsd_cmp.

closed 2021-09-01 by donn