The-OpenROAD-Project/OpenLane #236: SYNTH_ADDER_TYPE=RCA also requires SYNTH_READ_BLACKBOX_LIB=1

antonblanchard, 2021-02-13:
Commit a421bf6689fd ("Add Ripple carry adder/Full adder mapping to yosys (#228)") adds `SYNTH_ADDER_TYPE`. I tried the `RCA` option but it fails with:

```
ERROR: Module `\sky130_fd_sc_hd__fa_1' referenced in module `\adder' in cell `$add$/openLANE_flow/designs/adder/src/adder.v:20$2.stage[0].FA' is not part of the design.
```

Adding access to the standard cells with `set ::env(SYNTH_READ_BLACKBOX_LIB) 1` fixes it. Since the design itself is not using standard cells this is a bit confusing.

The `FA` doesn't have this issue, I'm guessing because we do that mapping later on.

antonblanchard, 2021-02-13:
A simple example: [adder.tar.gz](https://github.com/efabless/openlane/files/5976324/adder.tar.gz)

agorararmard, 2021-02-15:
@antonblanchard: So, I guess that it's running earlier in the script to do the mapping before yosys/abc runs any mapping or optimizations on the designs, and so it forces all adders to be `RCA`, in which case the module that is passed to ABC have some already instantiated cells and so requires `set ::env(SYNTH_READ_BLACKBOX_LIB) 1` because there are manually instantiated cells in the passed Verilog. In which case, the correct solution would be to force `set ::env(SYNTH_READ_BLACKBOX_LIB) 1` regardless of the user given value, if `SYNTH_ADDER_TYPE` is set to `RCA`.

@Manarabdelaty: am I correct in what I said above? ^

Manarabdelaty, 2021-02-15:
@agorararmard yes, that is correct.

agorararmard, 2021-02-15:
@antonblanchard: Do you mind PRing this simple adder to our `designs/` so that I'd include it in the CI tests in the future (don't forget to add a license header)?

closed 2021-02-15 by ax3ghazy