The-OpenROAD-Project/OpenLane #278: Get stuck at Global routing when implementing FPGA fabrics

nguyendao-uom, 2021-04-01:
**Describe the bug**
It's weird that the openlane flow can successfully run (RTL to GDS) for a fabric with 2x2 CLBs but get stuck at Global routing when implementing for a 4x4 CLBs
**To Reproduce**
Please find more details at https://github.com/FPGA-Research-Manchester/FABulous-SKY130

agorararmard, 2021-04-05:
@nguyendao-uom: I think you need to reduce the core utilization [here](https://github.com/FPGA-Research-Manchester/FABulous-SKY130/blob/ccd47dbd3ed653696a7a96df90932a7954e40be9/fabric4/sky130A_sky130_fd_sc_hdll_config.tcl#L4), also check [this](https://github.com/efabless/openlane/wiki#how-to-get-to-0-drc-violations).

nguyendao-uom, 2021-04-05:
Thanks @agorararmard. Actually I did change core utilization to 30 but didn't get through. Have you also tried to run the flow and pass it?

agorararmard, 2021-04-05:
@nguyendao-uom: This is because the library config file is called hdll instead of hd. https://github.com/FPGA-Research-Manchester/FABulous-SKY130/blob/main/fabric4/sky130A_sky130_fd_sc_hdll_config.tcl

nguyendao-uom, 2021-04-06:
@agorararmard : Yes, but it doesn't help after I changed the file name and the `CORE_UTIL` to 30 (shown in the config.tcl under runs directory).
Any other suggestions? Thanks

maliberty, 2021-04-28:
I cloned your repo and ran '/flow.tcl -design fabric4' but didn't see the GR error.  How do you reproduce it?

nguyendao-uom, 2021-04-29:
@maliberty it's weird. what openlane's version are you using? My one is efabless/openlane:rc6 and it's hanging at
....
Notice 0: Reading LEF file:  /openLANE_flow/designs/fabric4/runs/29-04_07-34/tmp/merged_unpadded.lef
Notice 0:     Created 13 technology layers
Notice 0:     Created 25 technology vias
Notice 0:     Created 438 library cells
Notice 0: Finished LEF file:  /openLANE_flow/designs/fabric4/runs/29-04_07-34/tmp/merged_unpadded.lef
Notice 0: 
Reading DEF file: /openLANE_flow/designs/fabric4/runs/29-04_07-34/results/placement/fabric4.placement.def
Notice 0: Design: fabric4
Notice 0:     Created 419 pins.
Notice 0:     Created 77425 components and 469902 component-terminals.
Notice 0:     Created 2 special nets and 0 connections.
Notice 0:     Created 61097 nets and 183188 connections.
Notice 0: Finished DEF file: /openLANE_flow/designs/fabric4/runs/29-04_07-34/results/placement/fabric4.placement.def

maliberty, 2021-04-29:
I just used the head of master (yesterday) in openlane and whatever version of OR it uses.

maliberty, 2021-04-29:
Fwiw it seems strange to throw an fpga fabric into a general purpose P&R flow.  The fabric should have a very natural row/column floorplan which the global placer is unlikely to reproduce.

nguyendao-uom, 2021-04-29:
@maliberty Except the timing for combinational loops in the fabric, I'm not sure why we can only run the openlane for tiny fabrics (from 2x2 to 3x4 CLBs) but not for 4x4 fabrics? But again, how you got it through with the 4x4 fabrics? If possible can you also try will our 15x15 fabric here https://github.com/FPGA-Research-Manchester/eFPGA---RTL-to-GDS-with-SKY130/tree/main/Verilog

shalan, 2021-04-29:
@nguyendao-uom I agree with @maliberty. Earlier we hardened an open source 8x8 embedded FPGA fabric using OpenLANE/OpenROAD. We had to manually floorplan the available die area into rows and columns. The placement of the building blocks macros was scripted. It worked fine and we were able to tape it out for Google's Open MPW1.

donn, 2021-09-29:
Stale, closed. Please create a new issue if this persists.

closed 2021-09-29 by donn