The-OpenROAD-Project/OpenLane #292: Feature Request: vhdl support

dhombios, 2021-04-12:
Having an open source toolchain for synthesizing integrated circuits can be really useful for research projects. However, some industries enforce the use of vhdl, which is not highly supported by open source projects (the only open source project with similar capabilities that supports vhdl is Electric EDA). Adding vhdl support to this project will allow universities and engineering research centers to reduce dependency on commercial tools and, therefore, making research more accessible. 

As Openlane is based on Yosis, this could be achievable using the GHDL plugin available for it, which adds support for that language and is also open source. Alternatively, Icarus Verilog provides a vhdl to verilog translator.

pylorak, 2021-05-24:
I'll add, VHDL is a lot more popular in Europe and is by far the dominant HDL language there (here :) ), taught both at universities and used in the industry.

donnie-j, 2022-03-27:
### What is this?
This is a patch that adds mixed language support for VHDL using GHDL though the Yosys GHDL module.

I adds these new environment variables, trying to match the style of OpenLane (for better or worse):

- SYNTH_BIN_OPTIONS Extra command line options for Yosys.  Put "-mghdl" in here
- VHDL_FILES Space separated list of files that the synthesis script will cause ghdl to analyze
- VHDL_MODULE_IMPORTS Modules Yosys will import from GHDL

### Usage
There are 2 ways to use this new capability, which you can also mix and match:

- Analyze with GHDL (and even for example simulate or target FPGA with) your VHDL sources first, and just leave VHDL_FILES (and if you wish, VERILOG_FILES) empty.  Then add the top level module to VHDL_MODULE_IMPORTS.
-  Put VHDL files you want OpenLane to analyze in VHDL_FILES.  This could be an OpenLane target specific top-level in which case analyze your other RTL units first as above, or all of your RTL.

### Caveats 
VHDL is a more complex language than Verilog.  It allows for record types, enumeration types, etc that cannot be represented in Verilog.  GHDL handles this by mostly flattening types and ports to simple vectors of wires.  The results are pretty useful and intuitive, but make sure you understand what it has done especially with respect to top level ports.

Everything works in practice AFAICS, except for escaping VHDL names for LVS.  The reason for this is Netgen produces output in JSON, and the OpenLane/openroad scripts use a 3rd party JSON parser in python that throws an exception on names such as \db.o.a[31]  The LVS has already happened, but OpenLane exits with an error... workaround: run with RUN_LVS = 0 until you are done working on the module, then rerun with RUN_LVS = 1 to be sure, and read the logs.

### Example OpenLane Project
On request.  It's pretty simple.  This has been run with projects as complex as the J2 SuperH compatible CPU core
[VHDL.diff.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8356782/VHDL.diff.gz)
.

mithro, 2022-03-28:
FYI -- @umarcor @antonblanchard

mithro, 2022-03-28:
I believe that @antonblanchard has been using VHDL heavily with OpenLane / OpenROAD for taping out https://github.com/antonblanchard/microwatt

antonblanchard, 2022-03-30:
This is great! One thing I've struggled with is how to power macros. Right now I modify the generated verilog with a script:

https://github.com/antonblanchard/microwatt/tree/caravel-mpw5-20220323/caravel

But this was developed over a year ago and perhaps the updates to the PDN code in Openlane and OpenROAD render it unnecessary now.