The-OpenROAD-Project/OpenLane #297: Issues encountered when installing v0.12 branch

IanBoyanZhang, 2021-04-23:
**Describe the bug**
Trying to follow the quickstart guide to install the toolchain. 

Hitting below error when `make pdk`

```
mkdir -p /home/pdks/open_pdks/sky130/sky130A/libs.tech/xschem
mkdir -p /home/pdks/open_pdks/sky130/sky130A/libs.tech/xschem
# Copy the entire repository (other than .git, if it exists)
if test "x../sources/xschem_sky130" != "x" ; then \
    cp -rp ../sources/xschem_sky130/* /home/pdks/open_pdks/sky130/sky130A/libs.tech/xschem ; \
fi
cp: cannot stat '../sources/xschem_sky130/*': No such file or directory
make[1]: *** [xschem-a] Error 1
make[1]: Leaving directory `/home/pdks/open_pdks/sky130'
make: *** [all-a] Error 2
```

**To Reproduce**
Steps to reproduce the behavior:

```
    git clone https://github.com/efabless/openlane.git --branch v0.12
    cd openlane/
    # Default PDK_ROOT is $(pwd)/pdks. If you want to install the PDK at a differnt location, uncomment the next line.
    #export PDK_ROOT=<absolute path to where skywater-pdk and open_pdks will reside>
    make openlane
    make pdk
```

**Expected behavior**

N/A

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04
 - Openlane Version [e.g. v0.12]
 - open_pdks Version: [b9ffc1f]
 - skywater-pdk Version: [commit hash]

donn, 2021-05-04:
Please use v0.15, our latest stable version.

ahmednofal, 2021-05-04:
This is also reproducible in develop branch as well as the v0.15 version.

donn, 2021-05-06:
So, I'll be honest, reproducing this issue has been a challenge for us. It seems to happen on some OS configurations but not others, for example, @ahmednofal had a similar issue but upon switching to Ubuntu 20.04 (he was on Debian), the issue disappeared.

@IanBoyanZhang We'd really appreciate your help in trying to isolate this issue. Can you share the output of the following commands:

* `lsb_release -a`
* `apt list --installed`

IanBoyanZhang, 2021-05-08:
- lsb_release -a

```
LSB Version:	core-9.20170808ubuntu1-noarch:security-9.20170808ubuntu1-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.5 LTS
Release:	18.04
Codename:	bionic
```
Is there a way I could share installed apt list privately?

donn, 2021-05-08:
Email me at mohamed.gaber@efabless.com.

IanBoyanZhang, 2021-05-09:
Sent. Thank you!

donn, 2021-06-22:
Sorry for the late reply, Ian- we traced this issue to a specific failure in open_pdks.

Namely, one of the files responsible for downloads fails silently. When it fails, critical files remain undownloaded, and then it can't find things like, well, xschem.

Patch has been submitted: https://github.com/RTimothyEdwards/open_pdks/pull/132

IanBoyanZhang, 2021-06-29:
Hi @donn, thanks for taking time to investigate and fix it!

donn, 2021-08-19:
No problems. This issue appears to be gone now, so I'll go ahead and close this.

closed 2021-08-19 by donn