The-OpenROAD-Project/OpenLane #298: OpenSTA unable to finish

cgreen18, 2021-04-23:
**Describe the bug**

Running flow on the provided test design, spm, works as intended.
When running the flow on a much larger design, the flow fails to finish synthesis. The flow successfully finished Yosys but fails during the scripts/sta.tcl portion of the flow. Specifically, the flow successfully reads scripts/base.sdc but fails on report_tns (line 28 of sta.tcl). It also fails/never completes when calling report_wns (line 30 of sta.tcl). The debug information is uninformative.

**To Reproduce**

The error is encountered for my team's ASIC verilog file. The flow is run on a the source code file: top_level_bASIC.generic.v     The verilog module is named "top_level_bASIC" and so is the design name.
Unknown how to reproduce for other designs/source code.

**Expected behavior**

Expected to finish flow. It freezes until "Child killed" see below.

**Screenshots**

Command:
./flow.tcl -design top_level_bASIC

The exact terminal output from the above command is given below. Note, the first portion is some output from reading base.sdc.

set_driving_cell -lib_cell $::env(SYNTH_DRIVING_CELL) -pin $::env(SYNTH_DRIVING_CELL_PIN) [all_inputs]
set cap_load [expr $::env(SYNTH_CAP_LOAD) / 1000.0]
puts "\[INFO\]: Setting load to: $cap_load"
[INFO]: Setting load to: 0.01765
set_load $cap_load [all_outputs]
[ERROR]: during executing: "sta /openLANE_flow/scripts/sta.tcl |& tee >&@stdout /openLANE_flow/designs/top_level_bASIC/runs/22-04_00-17/logs/synthesis/2-opensta"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: kill signal

[ERROR]: Please check sta log file
[ERROR]: Dumping to /openLANE_flow/designs/top_level_bASIC/runs/22-04_00-17/error.log
[INFO]: Calculating Runtime From the Start...
[INFO]: Flow failed for top_level_bASIC/22-04_00-17 in 0h53m39s
[INFO]: Generating Final Summary Report...
[INFO]: Design Name: top_level_bASIC
Run Directory: /openLANE_flow/designs/top_level_bASIC/runs/22-04_00-17
Source not found.
----------------------------------------

LVS Summary:
Source: /openLANE_flow/designs/top_level_bASIC/runs/22-04_00-17/results/lvs/top_level_bASIC.lvs_parsed.gds.log
Source not found.
----------------------------------------

Antenna Summary:
No antenna report found.
[INFO]: check full report here: /openLANE_flow/designs/top_level_bASIC/runs/22-04_00-17/reports/final_summary_report.csv
[ERROR]: Flow Failed.

while executing
"try_catch sta $::env(SCRIPTS_DIR)/sta.tcl |& tee $::env(TERMINAL_OUTPUT) [index_file $::env(opensta_log_file_tag) 0]"
(procedure "run_sta" line 8)
invoked from within
"run_sta"
(procedure "run_synthesis" line 12)
invoked from within
"run_synthesis"
(procedure "run_non_interactive_mode" line 14)
invoked from within
"run_non_interactive_mode {*}$argv"
invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
puts_info "Running interactively"
if { [info exists arg_values(-file)..."
(file "./flow.tcl" line 223)

**Desktop (please complete the following information):**
 - OS: CentOS-7 running w/ 4GB RAM and sufficient disk sapce
 - Openlane Versions rc4, rc7, and v0.12. hash: bee14bb44091f613ec226a4b399f30e1de5b817b
 - open_pdks Version hash: b9ffc1fd1cfc26cbca85a61c287ac799721f6e6a
 - skywater-pdk Version hash: db2e06709dc3d876aa6b74a5f3893fa5f1bc2a6e

**Additional context**
The source code is a generic netlist generated by a separate process.

donn, 2021-05-04:
Please provide the STA log file.

cgreen18, 2021-05-05:
Okay, I have uploaded the error logs: 2-opensta, error.log, and final_summary_report.
csv
The 2-opensta was turned into a .txt to upload to Github
[2-opensta.txt](https://github.com/efabless/openlane/files/6428809/2-opensta.txt)
[error.log](https://github.com/efabless/openlane/files/6428810/error.log)
[final_summary_report.csv](https://github.com/efabless/openlane/files/6428811/final_summary_report.csv)

donn, 2021-08-30:
Sorry for the late reply- can you share the design files, if you still have them? OpenLane has had a number of updates since then so the logs are not enough to isolate the issue anymore.

donn, 2021-11-23:
No reply for 60+ days: assuming this one is resolved.

closed 2021-11-23 by donn