The-OpenROAD-Project/OpenLane #303: ROUTING_OPT_ITERS no longer works

mattvenn, 2021-04-28:
**Describe the bug**

set ::env(ROUTING_OPT_ITERS) 100

should set the number of routing iterations to 100. Instead it is stuck at 64.

**To Reproduce**

On a congested design that takes longer than 64 iterations to route, increase iterations to a larger number:

set ::env(ROUTING_OPT_ITERS) 100

**Expected behavior**

router should keep going past 64th iteration

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 - Openlane Version 0.12
 - open_pdks Version:f788cea 
 - skywater-pdk Version: db2e067

mattvenn, 2021-04-28:
In case it's useful here's a screenshot after I set global iterations to 80

[image]

but still exited after 64

Manarabdelaty, 2021-04-28:
@mattvenn Can you share your design and config.tcl ? We think tritonroute either ignores the parameter or doesn't honour it if the violations are unsolvable. 

On another note, if you are still facing violations after the 64th iteration, you can try to slightly increase the `GLB_RT_ADJUSTMENT`.

mattvenn, 2021-04-29:
how would you prefer me to share the design? It is user_project_wrapper, so
it's part of caravel_user_project.

jaquerinte, 2021-05-13:
I have the same issue. I tested with lower size and with lower size of 64 works ( for example I tested with 20 and only does 20)

maliberty, 2021-07-21:
TritonRoute only does 64 iterations.  The control can limit it to less but can't force it higher.

donn, 2021-07-28:
OR team has clarified that this is intentional. We should maybe add a warning.

closed 2021-11-23 by donn