The-OpenROAD-Project/OpenLane #320: Not able to install PDK on Mac M1

guris12, 2021-05-04:
**Describe the bug**
chmod a+x /Users/gursimransodhi/Documents/OpenLane/openlane/pdk/skywater-pdk/env/downloads/Miniconda3-latest-MacOSX-arm64.sh
/Users/gursimransodhi/Documents/OpenLane/openlane/pdk/skywater-pdk/env/downloads/Miniconda3-latest-MacOSX-arm64.sh -p /Users/gursimransodhi/Documents/OpenLane/openlane/pdk/skywater-pdk/env/conda -b -f
touch "/Users/gursimransodhi/Documents/OpenLane/openlane/pdk/skywater-pdk/env/conda/bin/python"
touch: /Users/gursimransodhi/Documents/OpenLane/openlane/pdk/skywater-pdk/env/conda/bin/python: No such file or directory
make[1]: *** [/Users/gursimransodhi/Documents/OpenLane/openlane/pdk/skywater-pdk/env/conda/bin/python] Error 1
make: *** [skywater-library] Error 2

**To Reproduce**
Steps to reproduce the behavior:
git clone https://github.com/efabless/openlane.git
    cd openlane/
    make openlane
   
    make pdk
--> Getting Error 

**Expected behavior**
Should have installed PDK 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Mac M1 chip 
 - Openlane Version Latest
 - open_pdks Version: Latest
 - skywater-pdk Version: Latest
[image]

**Additional context**
Add any other context about the problem here.

donn, 2021-05-04:
Hi Guris, we're going to need a full log so we can help you here. The last 5 lines are not sufficient.

guris12, 2021-05-04:
hello donn, is there any log file which is generated ? I am not able to find it , could you please point me  to the right place

donn, 2021-05-04:
Just copy & paste more lines from the terminal.

donn, 2021-05-04:
Or alternatively, run:

  make pdk 2>&1 > ./make_pdk.log

Then find a file called make_pdk.log.

guris12, 2021-05-04:
[4may_error.log](https://github.com/efabless/openlane/files/6420154/4may_error.log)

please find the error.lo attached

donn, 2021-05-04:
…Might be me, but I don't see any errors in this log. Have you added the `2>&1`?

guris12, 2021-05-04:
Dumping a new file with above command

guris12, 2021-05-04:
please find the new log file 
[make_pdk.log](https://github.com/efabless/openlane/files/6421180/make_pdk.log)

closed 2022-01-03 by donn