The-OpenROAD-Project/OpenLane #321: Error during execution of detailed placement

Geetima2021, 2021-05-04:
[image]
I have been using the run_placement command but I am facing the above issue. I had simultaneously run the run_placement command a no, of times. Each time Warning: Site check failed (133) increase and also the no. of psn_inst_psn_buff_1301
increases but the error remains. I had also set the FP_CORE_UTIL to default 50, PL_TARGET_DENSITY to 0.55 and CELL_PAD 4. Kindly assist me in resolving the above issue as I am not being able to figure out anything.

guris12, 2021-05-04:
could you check log file ?

Geetima2021, 2021-05-04:
Yes I checked...This is what the error.log file shows.
during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_opendp.tcl |& tee >&@stdout /openLANE_flow/designs/picorv32a/runs/PD_03May/logs/placement/opendp.log"
Last 10 lines:
child process exited abnormally

Manarabdelaty, 2021-05-05:
@Geetima2021 Can you post the `opendp.log` file ? It should be under `logs/placement`

Geetima2021, 2021-05-06:
[image]

Geetima2021, 2021-05-06:
I exactly don't know what happened but it is most probably working and not showing any error now.

commit 31a60b9a (2021-05-21): fixes #321

set_if_unset apparently does not work with ::env(VAR). This passed the CI because CI rebuilds the PDK (Which has ::env(STD_CELL_LIBRARY_CDL) set correctly in the PDK's config.tcl).
closed 2021-05-06 by Geetima2021