The-OpenROAD-Project/OpenLane #327: Yosys optimized away an entire design when SYNTH_NO_FLAT=0

KorlaMarch, 2021-05-06:
**Describe the bug**
I am running into an issue with Yosys optimizing away my entire design (a modified version of `user_proj_example` from https://github.com/efabless/caravel_user_project) in the synthesis step. However, the problem went away when I set `SYNTH_NO_FLAT=1`. I think this is related to the no-driver warning. For instance, `Warning: Wire user_proj_example.\wbs_dat_o [31] is used but has no driver`. But I am not too sure. Logfile from both cases might be helpful:

**Log Files**
Good Run (SYNTH_NO_FLAT=1): https://pastebin.com/pfeXFcky
Bad Run (SYNTH_NO_FLAT=0): https://pastebin.com/zDCW3iEs

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the code from [brown-chip/caravel_user_project](https://github.com/brown-chip/caravel_user_project)
2. Setup openlane and caravel
3. remove `set ::env(SYNTH_NO_FLAT) 1` from `openlane/user_proj_conv/runs/user_proj_conv/config.tcl`
4. Run `make user_proj_conv`
5. Check the error in `openlane/user_proj_conv/runs/user_proj_conv/logs/synthesis/1-yosys.log`

**Expected behavior**
`1-yosys.log` should show the synthesis ended with only one multiplexer without `set ::env(SYNTH_NO_FLAT) 1`
but show multiple gates and memories with `set ::env(SYNTH_NO_FLAT) 1`

**Desktop (please complete the following information):**
 - OS: ubuntu
 - Openlane Version: v0.15
 - open_pdks Version: f788cea
 - skywater-pdk Version: bb2f842

sjkelly, 2021-05-06:
I looked through the project a bit. This may be due to not all wires/regs being hooked up in your top level verilog module.

`SYNTH_NO_FLAT` is defined here:

https://github.com/efabless/openlane/blob/a95f56dac367c05bc48b4a001c419b58ec9648a4/scripts/synth.tcl#L240-L243

`yosys -p 'help flatten'`:
```
This pass flattens the design by replacing cells by their implementation. This
pass is very similar to the 'techmap' pass. The only difference is that this
pass is using the current design as mapping library.

Cells and/or modules with the 'keep_hierarchy' attribute set will not be
flattened by this command.
```

As I understand it, without `flatten` yosys will treat modules as discrete entities and ensure all their inputs/outputs are preserved.  With `flatten`, yosys will optimize through modules you define, which includes deleting wires and optimizing the logic flow. 

You should be able to isolate the issue by running the following in yosys, with [caravel's defines.v](https://github.com/efabless/caravel/blob/master/verilog/rtl/defines.v)
`yosys -s flatten_test.ys`

flatten_test.ys:
```
read_verilog defines.v user_proj_conv.v ;
hierarchy -check -top user_proj_conv
synth -top user_proj_conv -flatten
stat
```

By changing the `-flatten` flag in the `synth` step I can reproduce your good and bad results.

KorlaMarch, 2021-05-06:
Thanks! I use `flatten_test.ys` from @sjkelly and was able to find and fix the issue within my code. Looks like it is a problem within my design and it is working now. Will close the issue!

closed 2021-05-06 by KorlaMarch