The-OpenROAD-Project/OpenLane #362: make test failed with "STD_CELL_LIBRARY_CDL no such variable"

pwang7, 2021-05-19:
**Describe the bug**
when running make test, the test broke, with the error "STD_CELL_LIBRARY_CDL no such variable"

**To Reproduce**
Steps to reproduce the behavior:
1. make test

**Expected behavior**
make test should pass

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: ubuntu 20.04
 - Openlane Version: master branch
 - open_pdks Version: default
 - skywater-pdk Version: default

**Additional context**
[image]

Luo2017, 2021-05-20:
Hi，there! 
I have encountered the same problem while I'm trying to execute the command `make test` just like u on Ubuntu 20.04 and v0.16 of it. And I found that this variable is defined in ./scripts/tcl_commands/cvc.tcl:

[image]

I doubt whether the definition of this variable really works😶. And in v0.15, I found a different expression like this without using the variable:

[image]

So I modified this file to make it just look like v0.15(or simply replace it):

[image]

And it works.

[image]

Hope it is useful to this issue and the problem should be really solved as soon as possible.✨

Luo2017, 2021-05-21:
Hi  @ahmednofal, thanks a lot. 
I've reviewed the commit history of this part and i think your submission has truly solved the problem.

commit f7412929 (2021-05-21): Merge pull request #368 from ahmednofal/patch-4

fixes #362
closed 2021-05-21 by jeffdi