The-OpenROAD-Project/OpenLane #400: Error for make test

deepsita, 2021-05-28:
**Describe the bug**
The following error is obtained when make test is run after openlane installation.

**Screenshots**
The screenshots of errors are attached.
[image]
[image]

**Desktop:**
 - OS: ubuntu]
 - Openlane Version [e.g. v0.12]
 - open_pdks Version: [commit hash]
 - skywater-pdk Version: [commit hash]

donn, 2021-06-21:
Magic has been updated in later versions of OL: can you give it a shot?

donn, 2021-08-30:
Staled out, locked.