The-OpenROAD-Project/OpenLane #424: Add facility to package failed runs

maliberty, 2021-06-05:
In OpenROAD-flow-scripts the 'make issue' facility makes it easy to package up all necessary files to reproduce a problem in order to reproduce an issue.  OpenLane needs a similar capability.  See the on-going discussion in https://github.com/The-OpenROAD-Project/OpenROAD/issues/703.

donn, 2021-06-06:
I'll be honest, I have no idea what on earth is going on here: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/b549b82ff7a3c004a0117ebea2ef970f2d4c7207/flow/util/utils.mk#L57

The (unofficial) ruby script I made simply tries to extract the variables and files, then replace them with resolved values. Does this do something similar?

maliberty, 2021-06-07:
It produces a tgz with runme script that (1) sources a script that captures all the envars set by make and (2) invokes openroad on the proper tcl file to reproduce the step that is failing.  That tgz is a complete capture of the problem and is trivial for a developer to unpack and run.

donn, 2021-06-07:
Huh. I like the idea of the runner script being the one to set the environment variables. I'll look into it.