The-OpenROAD-Project/OpenLane #426: Error in routing stage

Mansi-VLSI-3, 2021-06-06:
**Describe the bug**
The following error is obtained while running the routing. It says the **design congestion is too high**. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
The snapshots of the error obtained: 

[image]

[image]

[image]

**Desktop (please complete the following information):**
 - OS: ubuntu

maliberty, 2021-06-06:
That message comes from the global router not the detailed router.  It's not clear that this is a bug and not just an over congested design that needs more routing area.

Mansi-VLSI-3, 2021-06-06:
> That message comes from the global router not the detailed router. It's not clear that this is a bug and not just an over congested design that needs more routing area.

Hi, yes you are right the error came from the global router itself. Also tried it by reducing the initial utilization to 1% and 5% but eventually the **same error: design congestion is too high**  is obtained.

**For 1% utilization:** 
Following error is obtained while running the detailed_placement :
[image]

and the following error at the run_routing ( while running global routing ) :
[image]

**For 5% utilization:**
Following error is obtained in the run_routing stage ( while running global routing) :
[image]

maliberty, 2021-06-07:
Standalone opendp was integrated into openroad over a year ago is not being developed anymore.  OpenLane should move forward.

donn, 2021-06-07:
Will do

donn, 2021-08-19:
You can probably try now.

donn, 2021-08-30:
Yes, please always use the latest OpenLane.

vijayank88, 2021-09-01:
@Mansi-VLSI-3 possible share your config.tcl and source RTL. its easy to debug.

Mansi-VLSI-3, 2021-09-05:
> Yes, please always use the latest OpenLane.

Hi  , am already using the latest one that is docker rc6 so is there an update even after that because the issue still persists?

donn, 2021-09-05:
The latest tag is 2021.09.05_03.29.19.

rc6 is from December 2020.

closed 2021-11-23 by donn

vijayank88, 2023-04-12:
@18341A04A6 
Goto https://github.com/The-OpenROAD-Project/OpenLane/issues, start new issue and follow the guidelines to file an issue.
With screenshot it will not get resolved.