The-OpenROAD-Project/OpenLane #440: Make publications pdf or accessible link available

dralabeing, 2021-06-11:
I am unable to access the following publications cited in the docs:
1. M. Shalan and T. Edwards, “Building OpenLANE: A 130nm OpenROAD-based Tapeout- Proven Flow : Invited Paper,” 2020 IEEE/ACM International Conference On Computer Aided Design (ICCAD), San Diego, CA, USA, 2020, pp. 1-6.

2. R. Timothy Edwards, M. Shalan and M. Kassem, “Real Silicon using Open Source EDA,” in IEEE Design & Test, doi: 10.1109/MDAT.2021.3050000.**Describe the bug**
A clear and concise description of what the bug is.

Please provide pdfs.. thanks

donn, 2021-06-16:
I'm fairly sure that furnishing this request violates a copyright law of some kind. @mkkassem @shalan Care to weigh in?

dralabeing, 2021-06-24:
If this is not publishable please remove reference in the document.

donn, 2021-06-24:
It's already been published, but we may not be able to provide a copy as part of the repository lest we run afoul of some copyright law.

The references will remain there should someone feel the need to cite OpenLane in their research projects. Getting access to the papers is a separate matter.

dralabeing, 2021-06-25:
Ok sure. thanks.

closed 2021-06-25 by dralabeing

mithro, 2021-06-27:
You could link to the IEEE site like https://ieeexplore.ieee.org/document/9336682