The-OpenROAD-Project/OpenLane #450: LVS fails with fake diodes & MAGIC_EXT_USE_GDS

mattvenn, 2021-06-14:
**Describe the bug**

If LVS is run against the GDS (instead of DEF/LEF) by setting MAGIC_EXT_USE_GDS then LVS will fail if there are fake diodes in the design. Fake diodes are used if DIODE_INSERTION_STRATEGY is set to 5.

**To Reproduce**
Steps to reproduce the behavior:

add these lines to designs/spm/config.tcl

set ::env(DIODE_INSERTION_STRATEGY) "5"
set ::env(MAGIC_EXT_USE_GDS) "1"

then run the flow:

./flow.tcl -design spm

design will fail with LVS issues

**Expected behavior**

design should pass LVS.

**Desktop (please complete the following information):**
 - OS: Ubuntu 20 LTS
 - Openlane Version 0.15

-ne skywater-pdk 
bb2f842ac8d1b750677ca25bc71fb312859edb82
-ne open_pdks 
7e29496eecf3ee8e1766f1b7f9441f97204d4735

RTimothyEdwards, 2021-06-14:
@Manarabdelaty :  FYI, the issue can be fixed in sky130A_setup.tcl.  The "fakediode" cell needs to be added to the list of cells to ignore.  I had thought that the "fakediode" cell had a diode in it and just removed the contact, but it appears that the cell has only the pin and no diode.  This causes it to be optimized out by magic's extraction, and then LVS fails because the verilog netlist has fakediode cells but the layout extracted netlist doesn't.

I'm updating sky130A_setup.tcl in open_pdks now to correct the problem there, but it will need to be updated for OpenLane.

donn, 2021-08-30:
@RTimothyEdwards has this been updated?

RTimothyEdwards, 2021-08-30:
@donn :  Yes, this was fixed a long time ago.

donn, 2021-08-30:
Guess that’s that then.

closed 2021-08-30 by donn