The-OpenROAD-Project/OpenLane #462: detailed_routing is taking exceptionally long time while executing.

madgo1991, 2021-06-16:
While running full flow on some designs the "detailed_routing" is taking very long time to complete. During, say 30th iteration, it takes 17 hours to complete but in later iterations say 40th it takes even 38 hours to complete. This is causing the process to continue for many days. Is it normal? Is there any way to speed up the process? 

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Openlane Version v0.9
 - open_pdks Version: [commit hash] 1.0.141
 - skywater-pdk Version: [commit hash] v0.0.0-308-gdb2e067

donn, 2021-06-16:
Depending on the design, the answer is yes, some designs can take days to route, especially if you have tight area requirements, but shouldn't really be the case if you're using the defaults given that our core utilization is 50%. That's not that tightly packed.

More information and relevant logs would be helpful.

maliberty, 2021-06-16:
I would describe it as unusual but not actionable without a test case.

madgo1991, 2021-06-21:
Hi @donn The core utilization is set at 10%, so the design is not tightly packed. I reran the full flow on the design and now it is at 51st iteration.  I am attaching the logs
[i2c_17Jun21.txt](https://github.com/The-OpenROAD-Project/OpenLane/files/6683777/i2c_17Jun21.txt)

maliberty, 2021-06-21:
Something is not right as the router is making no progress however without a test case it is hard to say much.

It doesn't look to be 10% utilized based on your log file:

utilization                37 %

madgo1991, 2021-06-21:
@maliberty The FP_CORE_UTIL is the parameter we have to set right? Because that is set to 10. Is there another parameter to set utilization?

maliberty, 2021-06-21:
I'm not the best to answer about OL but I do see what OR is reporting.

madgo1991, 2021-09-02:
This issue was resolved as the FP_CORE_UTIL parameter was not set in config.tcl but in some other file. WHen the FP_CORE_UTIL was set to 10 it ran within time.

closed 2022-01-03 by donn