The-OpenROAD-Project/OpenLane #473: Segmentation fault during or_floorplan -> read_lef

bjmuld, 2021-06-21:
**Describe the bug**
Openlane fails during initial floorplanning `or_floorplan.tcl::20 -> read_lef`  with message:
```
[ERROR]: during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_floorplan.tcl |& tee /dev/null /openLANE_flow/designs/sky130_hilas_TopLevelTestStructure/runs/openlane_test/logs/floorplan/2-verilog2def.openroad.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
Components of this program may be licensed under more restrictive licenses which must be honored.
Warning: /home/bjmuld/work/hilas/fastlane/pdks/sky130A_hilas/libs.ref/sky130_hilas_sc/lib/sky130_hilas_sc.lib line 31, default_operating_condition tt_025C_1v80 not found.
Notice 0: Reading LEF file:  /openLANE_flow/designs/sky130_hilas_TopLevelTestStructure/runs/openlane_test/tmp/merged_unpadded.lef
Notice 0:     Created 13 technology layers
Notice 0:     Created 25 technology vias
Notice 0:     Created 54 library cells
Notice 0: Finished LEF file:  /openLANE_flow/designs/sky130_hilas_TopLevelTestStructure/runs/openlane_test/tmp/merged_unpadded.lef
child killed: segmentation violation
```

**To Reproduce**
Steps to reproduce the behavior:

1. Get our top-level design: https://gitlab.com/um-ece/ftl-lab/hilas/fastlane/-/tree/hilas/designs/sky130_hilas_TopLevelTestStructure

2. Get our standard cells:  https://gitlab.com/um-ece/ftl-lab/hilas/cells

(1_alt+2_alt:  Clone and setup our fork of OpenLANE:  https://gitlab.com/um-ece/ftl-lab/hilas/fastlane )

Build design `sky130_hilas_TopLevelTestStructure` with openlane.

**Expected behavior**
I expect a failure within `or_floorplan.tcl`  (or `read_lef`) to produce some kind of error message rather than a segfault.
As-is, I am not sure how to modify the design to correct

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: ubuntu 20.04
 - Openlane Version: `v0.15`
 - open_pdks Version: `b9ffc1f`
 - skywater-pdk Version: `db2e067`

**Additional context**
I did see similar issues raised by others and did try variations of tcl parameters:
```
set ::env(FP_CORE_UTIL) 2
set ::env(PL_TARGET_DENSITY 0.03
set ::env(PL_SKIP_INITIAL_PLACEMENT) 1
#set ::env(SYNTH_STRATEGY) "1"
#set ::env(FP_PDN_VOFFSET) 4
#set ::env(FP_PDN_VPITCH) 15
#set ::env(FP_PDN_HOFFSET) 4
#set ::env(FP_PDN_HPITCH) 15
```
but with no resolution.

donn, 2021-06-21:
This is an underlying OpenROAD issue. We'll ping you once we've updated the version of OpenROAD we're using (perhaps within 10 days or so.)

bjmuld, 2021-06-21:
Thanks for the update!  Can you point me to a branch somewhere where a fix has been proposed?

donn, 2021-06-21:
https://github.com/donn/openlane/tree/or_migration but again, we're still debugging it.

pwhardy, 2021-07-06:
From testing it looks like the problem is read_lef. I was able to get the command to execute from inside the docker environment and verify it wrote the merged file. But it still fails when running "make test"

bjmuld, 2021-07-06:
@pwhardy to clarify: you're getting into the docker and starting the `openroad` binary, then getting an interpreter prompt and issuing the `read_lef <LEF_FILE>` command with the same LEF file and youre _not_ seeing openroad segfault?

pwhardy, 2021-07-06:
@bjmuld you are correct.

maliberty, 2021-07-14:
@donn with the OR update, is this resolved or not?

bjmuld, 2021-07-14:
I checked out the brach Donn referenced and experienced the same issue...

maliberty, 2021-07-14:
in that case a corresponding openroad issue should be created.

bjmuld, 2021-07-21:
My mistake.  I believe Donn's `or_migration` branch failed before floorplanning for a reason I don't remember.

donn, 2021-08-05:
So I set up Hilas and your design and unfortunately, I can't get it to synthesize without modifying the scl cell definitions manually, after which it fails to floor plan. I tried running `make` to no avail either: same issues. All I can offer is this small fix for Unix-likes:

```diff
diff --git a/scripts/magic_gen.py b/scripts/magic_gen.py
index 9fa1fec..560d552 100755
--- a/scripts/magic_gen.py
+++ b/scripts/magic_gen.py
@@ -1086,7 +1086,7 @@ def make_verilog():
             cc = f.read()
         cell_names.update({lf: re.findall(r'\n\s*MACRO\s+(\S+)', cc)})
 
-    template_loader = jinja2.FileSystemLoader(TEMPLATE_PATH)
+    template_loader = jinja2.FileSystemLoader(str(TEMPLATE_PATH))
     template_env = jinja2.Environment(
         trim_blocks=True,
         lstrip_blocks=True,
@@ -1158,7 +1158,7 @@ def make_lib():
             cc = f.read()
         cell_names.update({lf: re.findall(r'\n\s*MACRO\s+(\S+)', cc)})
 
-    template_loader = jinja2.FileSystemLoader(TEMPLATE_PATH)
+    template_loader = jinja2.FileSystemLoader(str(TEMPLATE_PATH))
     template_env = jinja2.Environment(
         trim_blocks=True,
         lstrip_blocks=True,
@@ -1275,7 +1275,7 @@ def make_lib_summary_md():
     template_env = jinja2.Environment(
         trim_blocks=True,
         lstrip_blocks=True,
-        loader=jinja2.FileSystemLoader(TEMPLATE_PATH),
+        loader=jinja2.FileSystemLoader(str(TEMPLATE_PATH)),
         autoescape=jinja2.select_autoescape(
             disabled_extensions=('.v',),
             default_for_string=True,
@@ -1323,7 +1323,7 @@ def make_cell_details_md():
     template_env = jinja2.Environment(
         trim_blocks=True,
         lstrip_blocks=True,
-        loader=jinja2.FileSystemLoader(TEMPLATE_PATH),
+        loader=jinja2.FileSystemLoader(str(TEMPLATE_PATH)),
         autoescape=jinja2.select_autoescape(
             disabled_extensions=('.v',),
             default_for_string=True,
```

bjmuld, 2021-08-05:
> I can't get it to synthesize without modifying the scl cell definitions manually

Thanks for investigating. What modifications did you have to make?
Did your floorplanning fail with the `child killed unexpectedly` /segfault message?

donn, 2021-08-05:
No, it simply couldn't find one of the designs.

--
What I had to modify is, from what it looks like, yosys just categorically does not like this kind of port declaration:

```verilog
module a(
     port_x[1],
     port_x[0]
```

I am unsure, but it appears that the same `\`ifdef` appeared twice in the document, once hosting the faulty declarations, and once as a more normal declaration. I simply removed the first as it was aliasing the second.