The-OpenROAD-Project/OpenLane #475: make regression_test failing while trying to generate html report

soumilheble, 2021-06-22:
**Describe the bug**
make regression_test failed while trying to generate html report from csv. jinja2 PackageLoader unable to locate csv2html package.

**To Reproduce**
Steps to reproduce the behavior:
1. Run 'make regression_test'
alternatively even the following might work
1. Run 'make mount'
2. Run 'python3 ./scripts/csv2html/csv2html.py -i dummy.csv -o dummy.html'

**Expected behavior**
Probably generate a pretty html version of the csv report.

**Screenshots**
```
Traceback (most recent call last):
  File "./scripts/csv2html/csv2html.py", line 36, in <module>
    loader=PackageLoader('csv2html', 'templates'),
  File "/usr/local/lib/python3.6/site-packages/jinja2/loaders.py", line 285, in __init__
    import_module(package_name)
  File "/usr/lib64/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/openLANE_flow/scripts/csv2html/csv2html.py", line 36, in <module>
    loader=PackageLoader('csv2html', 'templates'),
  File "/usr/local/lib/python3.6/site-packages/jinja2/loaders.py", line 310, in __init__
    f"The {package_name!r} package was not installed in a"
ValueError: The 'csv2html' package was not installed in a way that PackageLoader understands.
Exiting thread 0
Exiting thread 1
Exiting thread 2
Exiting thread 3
Exiting thread 4
Exiting thread 5
Exiting thread 6
Exiting thread 7
Exiting thread 8
Exiting thread 9
Exiting thread 10
Exiting thread 11
Exiting thread 12
Exiting thread 13
Exiting thread 14
Exiting thread 15
Exiting thread 16
Exiting thread 17
Exiting thread 18
Exiting thread 19
Exiting thread 20
Exiting thread 21
Exiting thread 22
Exiting thread 23
Exiting thread 24
Exiting thread 25
Exiting thread 26
Exiting thread 27
Exiting thread 28
Exiting thread 29
Exiting thread 30
Exiting thread 31
Traceback (most recent call last):
  File "run_designs.py", line 402, in <module>
    subprocess.check_output(csv2html_result_cmd.split())
  File "/usr/lib64/python3.6/subprocess.py", line 356, in check_output
    **kwargs).stdout
  File "/usr/lib64/python3.6/subprocess.py", line 438, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['python3', './scripts/csv2html/csv2html.py', '-i', './regression_results/TEST_SW_HD_20_06_2021_23_11//TEST_SW_HD_20_06_2021_23_11.csv', '-o', './regression_results/TEST_SW_HD_20_06_2021_23_11//TEST_SW_HD_20_06_2021_23_11.html']' returned non-zero exit status 1.
make: *** [Makefile:172: regression_test] Error 1
```

**Desktop (please complete the following information):**
 - OS: Neon 20.04 focal; KDE neon User Edition 5.22
 - Openlane Version: master; 2752c82543ba87b772e585c6a160ed0bacffb624
 - open_pdks Version: b9ffc1fd1cfc26cbca85a61c287ac799721f6e6a
 - skywater-pdk Version: db2e06709dc3d876aa6b74a5f3893fa5f1bc2a6e

donn, 2021-06-22:
Yeah, this is a confirmed issue, I'm running into it. I'm looking for fix. Annoying part is I don't know when this surfaced.

soumilheble, 2021-06-22:
I'm not yet sure how jinja2 works but from the changelog looks like they made changes to the PackageLoader.

https://jinja.palletsprojects.com/en/3.0.x/changes/
```
Version 3.0.0
Released 2021-05-11
* Use PEP 451 API to load templates with PackageLoader.
```

```
bash-4.2$ python3
Python 3.6.8 (default, Nov 16 2020, 16:55:22) 
[GCC 4.8.5 20150623 (Red Hat 4.8.5-44)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import jinja2 as jin2
>>> print(jin2.__version__)
3.0.1
```

donn, 2021-06-22:
Thanks for the heads up. What an odd change.

donn, 2021-06-22:
Will be fixed as part of #433. Thank you for the report and subsequent investigation, contributions like this are invaluable to us. :)

closed 2021-08-19 by donn