The-OpenROAD-Project/OpenLane #476: Value of Removing Not-Really-Short-Short-Flags

donn, 2021-06-24:
# **Is your feature request related to a problem? Please describe.**
The entire codebase, but perhaps most alarmingly, run_designs.py, uses non-standard 2 character+ short flags.

For example, the short flag for `--threads` is `-th`.

>1 character short flags are unorthodox and fairly unique to this codebase (some BSD/Apple software uses `-long-flag/-long_flag`, but never more than one character for the short versions,) which would be disorienting to most people.

GNU getopt, the de-facto standard, also only supports one character for short flags.

# **Describe the solution you'd like**
Let's maybe not do that. It's incredibly disorienting. There are options: we can use capital letters and small letters if we're running out of short flags, but the Makefile has `-dts`, `-dl`. It looks like it's setting two flags called `ts` and `l`, which is not the case, these are two entirely separate flags.

I'm updating the Makefile to not use the short flags regardless if we act on this or not.

--
I would like your opinions on this. I can't possibly be the only person this is standing out like a sore thumb to.

donn, 2021-12-16:
Done.

closed 2021-12-16 by donn

mithro, 2021-12-16:
@donn Great work here!