The-OpenROAD-Project/OpenLane #477: long run time for make pdk during magic database creation

dralabeing, 2021-06-24:
Doc reference for this setup: https://openlane.readthedocs.io/en/latest/index.html#quick-start

Set  up the skywater-pdk and openlane using these steps:

step 1:   git clone https://github.com/efabless/openlane.git
step 2:    cd openlane/
step 3:    make openlane
step 4:    make pdk

make pdk after around 3 mins proceeds to creating the magic database, log below:
-----------------
Creating magic generation script to generate magic database files.
Running magic to create magic database files.

------------------

This magic DB creation takes approx. between 2 hrs 45 mins to 3Hrs 10min to complete.
The subsequent flow test passes fine.
If this is how long it takes then it should be documented and reflected in the make pdk output messages.
This was confirmed by several configurations Desktop, VM and recent OL versions.

donn, 2021-06-30:
@RTimothyEdwards sorry for bothering you tim

vijayank88, 2021-08-17:
@dralabeing  current v0.23 download speed enhanced. make pdk completed < 30min and make full-pdk completed <40min. Check at your end and you can close this issue. Remember i have reported this issue. Thanks

donn, 2021-08-17:
Please don't use outdated versions of OpenLane for filing bugs. v0.23 is a couple months old now.

dralabeing, 2021-08-17:
Thanks @vijayank88 .. the docs still refer to the old link: https://openlane.readthedocs.io/en/latest/#updating-openlane . didn't realize I am using an older version

closed 2021-08-17 by dralabeing