The-OpenROAD-Project/OpenLane #482: Newer Versions of Netgen Changed Their Handling Of Disconnected Pins

donn, 2021-07-01:
**Describe the bug**
See RTimothyEdwards/netgen#21.

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04/WSL2
 - Openlane Version: ba695ba016b22b55820d56f9db5d0299dbbcb37e

donn, 2021-07-01:
I've mitigated this by disabling `blabla` and `usb_cdc_core` as part of our tests, until we get a resolution to this.

donn, 2021-07-01:
Confirmed bug. We're going to merge anyway while we await a fix.

RTimothyEdwards, 2021-07-02:
@donn :  I believe that the issue is in your scripts.  You (or someone) pulled the "count_lvs.py" script from qflow.  This was updated recently;  see https://github.com/RTimothyEdwards/qflow/blob/master/scripts/count_lvs.py.in lines 84 to 88 (most of which is a comment).

donn, 2021-07-02:
Gotcha... will have to pull this is in and test.

donn, 2021-07-03:
@RTimothyEdwards Key is... we already reflected that. 5217162915c16057a4d63cad04525530b319c34c

donn, 2021-07-03:
Relevant tests passed. Apparently the rebase fixed this. Oops. Sorry for bothering you.

closed 2021-07-03 by donn