The-OpenROAD-Project/OpenLane #484: Provide detailed instructions for makefile to support docker connection to XServer

dralabeing, 2021-07-05:
Provide clear and detailed instructions for running GUIs within the docker- specifically for Xserver support. Merely, setting the DISPLAY variable is not sufficient.  The docker cmd needs a -e variable, for e.g 
docker run -v /tmp/.X11-unix:/tmp/.X11-unix -e DISPLAY=unix:1 -it --rm -v $(OPENLANE_DIR):/openLANE_flow [...]
This is necessary for remote machine operation and will save first time users a lot of time

A useful link: https://l10nn.medium.com/running-x11-applications-with-docker-75133178d090

leochand101, 2021-07-26:
Any updates on this ? I realized this issue in my first build as well, whats the intended use case ? I was trying to open the built gds on klayout to find the Xserver error. 
Should I build from source or is docker file image the typical use case in which case we do need to have X11 support.

dralabeing, 2021-07-26:
@donn Any updates on this?

donn, 2021-07-26:
This is not an intended use case for OpenLane. If people would like to view their designs in Klayout, the recommended approach is to install Klayout on their machine. We only use Klayout/Magic/etc in batch mode.

maliberty, 2021-07-26:
@donn why not just add a few lines of documentation and enable this for users?  In ORFS we have make targets to run klayout on the result.

donn, 2021-07-26:
Not opposed to it if someone can spare the manpower to do so, but given that Klayout has been a particular trouble spot, I... do not have the mental energy to add and test a new set of options. :)

maliberty, 2021-07-26:
@dralabeing do you want to make a PR for this?

vijayank88, 2021-07-27:
> Any updates on this ? I realized this issue in my first build as well, whats the intended use case ? I was trying to open the built gds on klayout to find the Xserver error.
> Should I build from source or is docker file image the typical use case in which case we do need to have X11 support.

Try this docker command while mount:
docker run -v /tmp/.X11-unix:/tmp/.X11-unix -e DISPLAY=unix:0.0 -it --rm -v $(OPENLANE_DIR):/openLANE_flow [...]

vijayank88, 2021-07-27:
> Any updates on this ? I realized this issue in my first build as well, whats the intended use case ? I was trying to open the built gds on klayout to find the Xserver error.
> Should I build from source or is docker file image the typical use case in which case we do need to have X11 support.

I tried in my machine below format working fine. Use below command while mount openlane:
docker run -v /tmp/.X11-unix:/tmp/.X11-unix -e DISPLAY=unix:0.0 -it --rm -v $(OPENLANE_DIR):/openLANE_flow [...]

leochand101, 2021-07-31:
Thanks I will give it a try. I built openroad from source as well and considering that docker is for cross platform deployment, I dont need that yet.

donn, 2021-12-12:
Addressed in #740 I guess

closed 2021-12-12 by donn