The-OpenROAD-Project/OpenLane #491: openlane designs/151 exited with error

vijayank88, 2021-07-10:
during executing: "bash /openLANE_flow/scripts/klayout/xor.sh /openLANE_flow/designs/151/runs/08-07_06-36/results/magic/riscv_top.gds /openLANE_flow/designs/151/runs/08-07_06-36/results/klayout/riscv_top.gds riscv_top /openLANE_flow/designs/151/runs/08-07_06-36/results/klayout/riscv_top.xor.xml |& tee >&@stdout /openLANE_flow/designs/151/runs/08-07_06-36/logs/klayout/36-klayout.xor.log"
Last 10 lines:
child process exited abnormally

Error log attached for reference captured from terminal.
===========
Elapsed: 0.000s
"^" in: xor.drc:38
ERROR: In /openLANE_flow/scripts/klayout/xor.drc: std::bad_alloc in Region::^
Writing layout file: /openLANE_flow/designs/151/runs/08-07_10-34/results/klayout/riscv_top.xor.gds ..
ERROR: std::bad_alloc in Region::^ in MacroInterpreter::execute
  /openLANE_flow/scripts/klayout/xor.drc:38:in `block in execute_drc'
  /openLANE_flow/scripts/klayout/xor.drc:35:in `each'
  /openLANE_flow/scripts/klayout/xor.drc:35:in `execute_drc'
  :/built-in-macros/drc_interpreters.lym:18:in `instance_eval'
  :/built-in-macros/drc_interpreters.lym:18:in `execute_drc'
  :/built-in-macros/drc_interpreters.lym:92:in `execute'
[ERROR]: during executing: "bash /openLANE_flow/scripts/klayout/xor.sh /openLANE_flow/designs/151/runs/08-07_10-34/results/magic/riscv_top.gds /openLANE_flow/designs/151/runs/08-07_10-34/results/klayout/riscv_top.gds riscv_top /openLANE_flow/designs/151/runs/08-07_10-34/results/klayout/riscv_top.xor.gds |& tee >&@stdout /openLANE_flow/designs/151/runs/08-07_10-34/logs/klayout/34-klayout.xor.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

[ERROR]: Please check bash  log file
[ERROR]: Dumping to /openLANE_flow/designs/151/runs/08-07_10-34/error.log
[INFO]: Calculating Runtime From the Start...
[INFO]: Flow failed for riscv_top/08-07_10-34 in 1h41m21s
[INFO]: Generating Final Summary Report...
[INFO]: Design Name: riscv_top
Run Directory: /openLANE_flow/designs/151/runs/08-07_10-34
Source not found.
----------------------------------------

LVS Summary:
Source: /openLANE_flow/designs/151/runs/08-07_10-34/results/lvs/riscv_top.lvs_parsed.gds.log
Source not found.
----------------------------------------

Antenna Summary:
No antenna report found.
[INFO]: check full report here: /openLANE_flow/designs/151/runs/08-07_10-34/reports/final_summary_report.csv
[ERROR]: Flow Failed.

    while executing
"try_catch bash $::env(SCRIPTS_DIR)/klayout/xor.sh  $arg_values(-layout1) $arg_values(-layout2) $::env(DESIGN_NAME)  $arg_values(-output_gds)  |& tee $..."
    (procedure "run_klayout_gds_xor" line 20)
    invoked from within
"run_klayout_gds_xor"
    (procedure "run_non_interactive_mode" line 35)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	puts_info "Running interactively"
	if { [info exists arg_values(-file)..."
    (file "./flow.tcl" line 223)
=============

**To Reproduce**
./flow.tcl -design [151]

**Desktop (please complete the following information):**
 - OS: [e.g. ubuntu]
 - Openlane Version [e.g. v0.12]
 - open_pdks Version: [commit hash]
 - skywater-pdk Version: [commit hash]
 -

donn, 2021-07-10:
Yeah, Klayout tends to do that. We're looking into a fix.

maliberty, 2021-07-26:
@donn if you are looking into a fix I won't duplicate your effort.  Do you have an eta?

donn, 2021-07-26:
@maliberty Not yet. Likely would entail updating Klayout.

maliberty, 2021-07-26:
It looks like you are on 26.9.  27.x had a lot of improvements so you should definitely upgrade.  It should be as simple as updating your Dockerfile and running the tests.  Given the priority is that something you could try soon?

donn, 2021-07-26:
I'm having a lot of fun over at Docker/for-win#11693 atm, so give me like a couple more hours. I'm setting up Podman as an alternative at the moment.

donn, 2021-07-26:
Btw, as a workaround, you can change this line in configuration/general.tcl:

```tcl
set ::env(RUN_KLAYOUT_XOR) 1
```

to
```tcl
set ::env(RUN_KLAYOUT_XOR) 0
```

commit 6f95c7e5 (2021-07-26): Update Klayout

Fixes #491