The-OpenROAD-Project/OpenLane #492: Using Yosys supported SystemVerilog?

loverdec, 2021-07-21:
Hi,

I'm trying to run a design that includes SystemVerilog in the design, but only the Direct Programming Interface (DPI) which is supported by Yosys

According to the docs
"SystemVerilog to Verilog conversion must be performed by the user (e.g., using bsg sv2v or any tool of their choosing) before running Yosys."

Does this mean NO SystemVerilog is supported whatsoever? Or is there some alternative I can do?

Here is the code and error for reference.

```
import "DPI-C" function int serial_tick
(
    input  bit     serial_out_valid,
    output bit     serial_out_ready,
    input  int     serial_out_bits,

    output bit     serial_in_valid,
    input  bit     serial_in_ready,
    output int     serial_in_bits
);

module SimSerial (
    input         clock,
    input         reset,
    input         serial_out_valid,
    output        serial_out_ready,
    input  [31:0] serial_out_bits,

    output        serial_in_valid,
    input         serial_in_ready,
    output [31:0] serial_in_bits,

    output        exit
);

    bit __in_valid;
    bit __out_ready;
    int __in_bits;
    int __exit;

    reg __in_valid_reg;
    reg __out_ready_reg;
    reg [31:0] __in_bits_reg;
    reg __exit_reg;

    assign serial_in_valid  = __in_valid_reg;
    assign serial_in_bits   = __in_bits_reg;
    assign serial_out_ready = __out_ready_reg;
    assign exit = __exit_reg;

    // Evaluate the signals on the positive edge
    always @(posedge clock) begin
        if (reset) begin
            __in_valid = 0;
            __out_ready = 0;
            __exit = 0;

            __in_valid_reg <= 0;
            __in_bits_reg <= 0;
            __out_ready_reg <= 0;
            __exit_reg <= 0;
        end else begin
            __exit = serial_tick(
                serial_out_valid,
                __out_ready,
                serial_out_bits,
                __in_valid,
                serial_in_ready,
                __in_bits
            );

            __out_ready_reg <= __out_ready;
            __in_valid_reg  <= __in_valid;
            __in_bits_reg   <= __in_bits;
            __exit_reg <= __exit[0];
        end
    end

endmodule
```

3. Executing Verilog-2005 frontend: ...

... SimSerial.v:3: ERROR: syntax error, unexpected TOK_ID, expecting ',' or ')'

maliberty, 2021-07-21:
I think people use sv2v as a workaround.

ChristianB12, 2021-07-21:
I tried using it, but it doesn't seem to support DPI-C

https://github.com/zachjs/sv2v

mithro, 2021-07-21:
You might want to consider [Surelog](https://github.com/chipsalliance/Surelog) + [UHDM](https://github.com/chipsalliance/uhdm) as an option for [parsing SystemVerilog for usage with Yosys](https://antmicro.com/blog/2020/12/ibex-support-in-verilator-yosys-via-uhdm-surelog/). The team at @antmicro has recently been focused on making the Surelog+UHDM -> Yosys -> OpenROAD flow more seamlessly usable -- might want to check out my diagram at https://j.mp/sv-flow-diagram too.