The-OpenROAD-Project/OpenLane #493: save_state crash when saving environment variables that contain double quotes

getziadz, 2021-07-21:
**Description**
OpenLane crashes when running the example design.
The source of the crash is the save_state function which crashes when it encounters environment variables with double quotes

**To Reproduce**
Define an environment variable that contains double quotes, such as:
`function urg() { command urg -full64 "$@"; }; export -f urg`

run
`make test`

You will observe the following:
```
[INFO]: Saving Runtime Environment
extra characters after close-quote
    while executing
"set ::env(BASH_FUNC_urg()) "() {  command urg -full64 "$@"
}""
    ("uplevel" body line 1)
    invoked from within
"uplevel #0 ${cmd}"
    (procedure "set_log" line 3)
    invoked from within
"set_log ::env($index) $::env($index) $::env(GLB_CFG_FILE) 1"
    (procedure "save_state" line 8)
    invoked from within
"save_state"
    (procedure "run_non_interactive_mode" line 69)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
        puts_info "Running interactively"
        if { [info exists arg_values(-file)..."
    (file "./flow.tcl" line 346)
```

**Expected behavior**
OpenLane crashes

**Desktop (please complete the following information):**
 - OS: Springdale Linux release 7.9 (Verona)
 - Openlane Version:
   - v0.22-3-g7b3af66
   - master branch ( 02c16938aea3770c1d2e03fc8beed763fa83f30f )

**Additional context**
A simple solution is to escape the double quotes. See https://github.com/getziadz/OpenLane/commit/ec2d52f5df8d7fe30f04323931b4be71f235fc8e

donn, 2021-07-26:
Thank you so much @getziadz- I'll merge the fix within the next few days.

getziadz, 2021-07-26:
No problem. Should I create a pull request?

donn, 2021-07-26:
That would greatly accelerate things, yes.

getziadz, 2021-07-26:
Just created the pull request.

donn, 2021-07-27:
Thank you so much.

ahmednofal, 2021-09-28:
I believe this to be still persistent in version 2021.09.19_20.25.16  
This is the traceback from the crash.
```
invalid command name ""/home/nofal/design/project_using_sky130A/verilog/source/spm.v""
while executing
"\"/home/nofal/design/project_using_sky130A/verilog/source/spm.v\""
("uplevel" body line 1)
invoked from within
"uplevel #0 ${cmd}"
(procedure "set_log" line 3)
invoked from within
"set_log ::env($index) $escaped_env_var $::env(GLB_CFG_FILE) 1"
(procedure "save_state" line 9)
invoked from within
"save_state"
(procedure "run_non_interactive_mode" line 68)
invoked from within
"run_non_interactive_mode {*}$argv"
invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
puts_info "Running interactively"
if { [info exists arg_values(-file)..."
(file "/OpenLane/flow.tcl" line 354)
```
It is worth noting that I am using a json configuration file instead of the regular tcl file.
Contents of the config.json
```
{ "CLOCK_NET": "clk", "CLOCK_PERIOD": "10", "CLOCK_PORT": "clk", "CLOCK_WIRE_RC_LAYER": "met5", "DATA_WIRE_RC_LAYER": "met2", "DESIGN_NAME": "spm", "PDK": "sky130A", "PDK_ROOT": "/home/nofa/os_eda/pdks", "PDK_VARIANT": "sky130_fd_sc_hd", "STD_CELL_LIBRARY": "sky130_fd_sc_hd", "VERILOG_FILES": [ "/home/nofal/design/project_using_sky130A/verilog/source/spm.v" ] }
```

donn, 2021-12-12:
Think this should be fine now, I just made all variables use `{}`.

If you have a variable with `{}` that's your problem.

closed 2021-12-12 by donn