The-OpenROAD-Project/OpenLane #494: make full-pdk download failed

vijayank88, 2021-07-22:
**Describe the bug**
For PDK download two options given. **make pdk** working fine. If i try to download **make full-pdk** failed after 1 hr.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'make full-pdk'
2. See error

Downloading alphanumeric library from https://github.com/PaulSchulz/sky130_pschulz_xx_hd
Cloning https://github.com/PaulSchulz/sky130_pschulz_xx_hd to ../sources/sky130_ml_xx_hd
Cloning into '../sources/sky130_ml_xx_hd'...
fatal: unable to look up current user in the passwd file: no such user
Unexpected end of command stream
make: *** [alpha-repo] Error 128
make: *** [build-pdk] Error 2

**Expected behavior**
successful download of full-pdk. 

**Desktop (please complete the following information):**
 - OS: [e.g. ubuntu]
 - Openlane Version [e.g. v0.12] trying to install v0.22
 - open_pdks Version: [commit hash]
 - skywater-pdk Version: [commit hash]

donn, 2021-07-26:
This is a git bug we've encountered previously. Can you please share your operating system?

vijayank88, 2021-07-27:
@donn I tried with Ubuntu 20.04.2 LTS and Centos 7 Linux 3.10.0-1160.36.2.el7.x86_64 both failed to download full_pdk
Thanks.,

donn, 2021-07-27:
Well, CentOS 7 will fail unless you manually install a later version of git, but Ubuntu 20.04.2 LTS should work. Can you give me the output of your `git --version`?

vijayank88, 2021-07-27:
> Well, CentOS 7 will fail unless you manually install a later version of git, but Ubuntu 20.04.2 LTS should work. Can you give me the output of your `git --version`?

git version 2.25.1

scveena, 2021-08-09:
Hi, When running make I encountered following error during getting the pdk. Can you please help?
Installing pip dependencies: - Ran pip subprocess with arguments:
['/Volumes/SeaGate/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/bin/python', '-m', 'pip', 'install', '-U', '-r', '/Volumes/SeaGate/pdks/skywater-pdk/condaenv.j46acple.requirements.txt']
Pip subprocess output:

Pip subprocess error:
ERROR: Exception:
Traceback (most recent call last):
  File "/Volumes/SeaGate/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_internal/cli/base_command.py", line 173, in _main
    status = self.run(options, args)
  File "/Volumes/SeaGate/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_internal/cli/req_command.py", line 203, in wrapper
    return func(self, options, args)
  File "/Volumes/SeaGate/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_internal/commands/install.py", line 286, in run
    reqs = self.get_requirements(args, options, finder, session)
  File "/Volumes/SeaGate/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_internal/cli/req_command.py", line 384, in get_requirements
    for parsed_req in parse_requirements(
  File "/Volumes/SeaGate/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_internal/req/req_file.py", line 135, in parse_requirements
    for parsed_line in parser.parse(filename, constraint):
  File "/Volumes/SeaGate/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_internal/req/req_file.py", line 319, in parse
    yield from self._parse_and_recurse(filename, constraint)
  File "/Volumes/SeaGate/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_internal/req/req_file.py", line 348, in _parse_and_recurse
    yield from self._parse_and_recurse(req_path, nested_constraint)
  File "/Volumes/SeaGate/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_internal/req/req_file.py", line 324, in _parse_and_recurse
    for line in self._parse_file(filename, constraint):
  File "/Volumes/SeaGate/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_internal/req/req_file.py", line 353, in _parse_file
    _, content = get_file_content(filename, self._session)
  File "/Volumes/SeaGate/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_internal/req/req_file.py", line 518, in get_file_content
    resp = session.get(url)
  File "/Volumes/SeaGate/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_vendor/requests/sessions.py", line 555, in get
    return self.request('GET', url, **kwargs)
  File "/Volumes/SeaGate/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_internal/network/session.py", line 454, in request
    return super().request(method, url, *args, **kwargs)
  File "/Volumes/SeaGate/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_vendor/requests/sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "/Volumes/SeaGate/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_vendor/requests/sessions.py", line 697, in send
    r.content
  File "/Volumes/SeaGate/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_vendor/requests/models.py", line 836, in content
    self._content = b''.join(self.iter_content(CONTENT_CHUNK_SIZE)) or b''
  File "/Volumes/SeaGate/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_vendor/requests/models.py", line 769, in generate
    chunk = self.raw.read(chunk_size)
AttributeError: 'FileNotFoundError' object has no attribute 'read'

failed

CondaEnvException: Pip failed

make[1]: *** [/Volumes/SeaGate/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/bin/python] Error 1
make: *** [skywater-library] Error 2

scveena, 2021-08-09:
Hi, I am trying to run the same flow explained in https://www.youtube.com/watch?v=d0hPdkYg5QI

Cheers
-Veena

donn, 2021-08-09:
@scveena Your issue is separate, and also resolved as of the latest patch.

If you’re following this video, ensure you clone `https://github.com/The-OpenROAD-Project/OpenLane`. The latest version of OpenLane solves this particular issue.

d-m-bailey, 2021-08-17:
@vijayank88 This is the same as #532. (You were first!) Should be fixed now with #534. If you test with the latest version and it works, could you close the issue?

vijayank88, 2021-08-17:
@d-m-bailey 
Thanks for the update. Now I'm able download full-pdk. In Makefile I was faced issue with python3. Replaced with python3.8 and able to download successfully.

closed 2021-08-17 by vijayank88