The-OpenROAD-Project/OpenLane #496: bug in or_floorplan.tcl

jayantcdsouza, 2021-07-23:
When running the openlane flow for design "spm" (and other designs). I see the following:

[ERROR STA-0402] initialize_floorplan -tracks is not a known keyword or flag.
Error: or_floorplan.tcl, 90 STA-0402
child process exited abnormally

The or_floorplan.tcl script has the following in it:
initialize_floorplan
-die_area $::env(DIE_AREA)
-core_area $::env(CORE_AREA)
-tracks $::env(TRACKS_INFO_FILE)
-site $::env(PLACE_SITE)

} else {

initialize_floorplan
-utilization $::env(FP_CORE_UTIL)
-aspect_ratio $::env(FP_ASPECT_RATIO)
-core_space "$bottom_margin $top_margin $left_margin $right_margin"
-tracks $::env(TRACKS_INFO_FILE)
-site $::env(PLACE_SITE)

However, the openroad floorplan tool does not have a "-tracks" switch:

% help initialize_floorplan
initialize_floorplan [-utilization util] [-aspect_ratio ratio]
[-core_space space | {bottom top left right}] [-die_area {lx ly ux uy}]
[-core_area {lx ly ux uy}] [-site site_name]

It looks like there might a different way to add tracks (either using make_tracks or some other command).

I did a complete clean download of the repo on July 22, 2021.

mdzakirhussain, 2021-07-24:
Yes, I too have the same issue.

zignig, 2021-07-25:
rolling back to v0.18 did not work 

 export IMAGE_NAME=efabless/openlane:v0.18

looks like a openlane issue !! :)

donn, 2021-07-26:
Ah. I recall… getting rid of this... and putting it behind a flag. Let me look into it.

donn, 2021-07-27:
So, uh, this is confusing:

https://github.com/The-OpenROAD-Project/OpenLane/blob/3c0f2fd0e5daf1ff1047fa802b7cc53bb8d4f7a7/scripts/openroad/or_floorplan.tcl#L48

The `-tracks` option has been removed from master for around a month now.

So uh, I'll need to know the command you're running exactly. and your `git log`.

jayantcdsouza, 2021-07-27:
I initially ran into this issue when running "make test". I subsequently looked into the scripts directory and found the -tracks switch.

Attached is my git log.
[jd_git.log](https://github.com/The-OpenROAD-Project/OpenLane/files/6887454/jd_git.log)

vijayank88, 2021-07-29:
yes this issue arise if you merge  v0.18 with v0.21. Backups results and Re-Install latest will works for me.

jayantcdsouza, 2021-08-04:
> yes this issue arise if you merge v0.18 with v0.21. Backups results and Re-Install with latest works for me.

Strange. I did not download v0.18 before downloading v0.21. I tried re-installing and still have the same problem.

vijayank88, 2021-08-09:
> > yes this issue arise if you merge v0.18 with v0.21. Backups results and Re-Install with latest works for me.
> 
> Strange. I did not download v0.18 before downloading v0.21. I tried re-installing and still have the same problem.

I'm using v0.23. I am not facing any issues. -tracks option not available in or_floorplan.tcl

jayantcdsouza, 2021-08-12:
I think the problem is if you pull using the following link:
 git clone https://github.com/efabless/openlane.git

If you pull using this link, the scripts downloaded are correct:
git clone https://github.com/The-OpenROAD-Project/OpenLane.git

However, now I get another issue:
Top-level design name: spm
Traceback (most recent call last):
  File "/openLANE_flow/scripts/write_powered_def.py", line 95, in <module>
    assert VDD_PORTS and GND_PORTS, "No power ports found at the top-level. "\
AssertionError: No power ports found at the top-level. Make sure that they exist and have the USE POWER|GROUND property or they match the arguments specified with --power-port and --ground-port.
child process exited abnormally

[ERROR]: Please check openroad  log file
[ERROR]: Dumping to /openLANE_flow/designs/spm/runs/openlane_test/error.log
[INFO]: Calculating Runtime From the Start...
[INFO]: Flow failed for spm/12-08_02-10 in 0h2m48s
[INFO]: Generating Final Summary Report...
[INFO]: Design Name: spm
Run Directory: /openLANE_flow/designs/spm/runs/openlane_test
Source not found.
----------------------------------------

donn, 2021-08-12:
You will need to re-build the PDK.

Again, the efabless fork is considerably out of date, that even the PDK has received updates in that time. I understand that this is an inconvenience.

Please invoke `make pdk` and report back.

jayantcdsouza, 2021-08-12:
thanks Donn! After updating the PDK, the flow works fine! Thanks for the tips!

donn, 2021-08-15:
Cheers, glad to help. :)

closed 2021-08-15 by donn