The-OpenROAD-Project/OpenLane #497: Routing stage failing- Blocking flow- finish opendb logger conversion #907

vijayank88, 2021-07-24:
**Describe the bug**
While i run eth_top design at routing stage get stopped with below error.
======
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 3 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: error, net 26404, shorts to another term
Notice 0: wire point (447385 896300)
Notice 0:  clknet_leaf_256_wb_clk_i

Error 0: order_wires failed
Notice 0: error, net 26404, shorts to another term
Notice 0: wire point (447385 896300)
Notice 0:  clknet_leaf_256_wb_clk_i

Error 0: order_wires failed
Notice 0: error, net 26404, shorts to another term
Notice 0: wire point (447385 896300)
Notice 0:  clknet_leaf_256_wb_clk_i

Error 0: order_wires failed
Notice 0: error, net 26404, shorts to another term
Notice 0: wire point (447385 896300)
Notice 0:  clknet_leaf_256_wb_clk_i

Error 0: order_wires failed
Notice 0: error, net 26404, shorts to another term
Notice 0: wire point (447385 896300)
Notice 0:  clknet_leaf_256_wb_clk_i

Error 0: order_wires failed
Notice 0: error, net 26404, shorts to another term
Notice 0: wire point (447385 896300)
Notice 0:  clknet_leaf_256_wb_clk_i

Error 0: order_wires failed
======

**To Reproduce**
Steps to reproduce the behavior:
1. Go to ' https://github.com/vijayank88/vijayank88/blob/main/eth_top.tar.xz '
2. Click on 'eth_top.tar.xz'
3. Scroll down to 'untar and goto eth_top/runs/ untar coreModify.tar.xz coreModify2/logs/routing/fastroute.log'
4. See fastroute.log end of the file for error message

**Expected behavior**
Expecting run_routing stage to complete successfully.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Centos 7]
 - Openlane Version [e.g. v0.22]
 - open_pdks Version: [Refer coreModify2 for all logs]
 - skywater-pdk Version: [Refer coreModify2 for all logs]

**Additional context**
I am able to run successfully in OpenROAD flow . If I add same design in openLane setup its getting stopped at routing stage

maliberty, 2021-07-26:
"To Reproduce" is a description of how to see the log file not how to reproduce the problem (ie run the tool to produce that result).

@dralabeing if "I am able to run successfully in OpenROAD flow" why is this an OR issue?

dralabeing, 2021-07-27:
@maliberty this is to address the Notice message which originates in OpenROAD, does it not?

maliberty, 2021-07-27:
@dralabeing "While i run the_top design at routing stage get stopped with below error."   Is this just about the formatting of the message or the failure?  I assumed the later.

dralabeing, 2021-07-27:
@maliberty This is both an error in routing through the OpenLane flow and the formatting as indicated by the report. Not sure, why the OR standalone flow passes successfully. 
@vijayank88 : Can you please clarify if the OR flow passed successully? Please provide the log file for this section of the test.

maliberty, 2021-07-27:
@dralabeing messaging is covered by https://github.com/The-OpenROAD-Project/OpenROAD/issues/907.

vijayank88, 2021-07-27:
@dralabeing 
OpenLane flow i used with sky130A library. It failing at routing stage with the message "Error 0: order_wires failed
Notice 0: error, net 26404, shorts to another term
Notice 0: wire point (447385 896300)
Notice 0: clknet_leaf_256_wb_clk_i"

In OpenRoad flow i used with nangate45 platform. Its completed successfully. Refer below links for logs.
https://github.com/vijayank88/vijayank88/blob/main/logs.tar.xz
Thanks.,

maliberty, 2021-07-27:
Running on a different pdk is completely different.  You should use either sky130hd or sky130hs (I'm not sure which library OL uses) in ORFS to reproduce the issue.

maliberty, 2021-07-27:
I'm not familiar enough with OL to figure out how to take your tar and turn it into a run of openroad that I can look at.  Would you or @donn provide steps.

vijayank88, 2021-07-28:
@maliberty 
Thanks for your time to debug this issue. In ORFS no issues to run my eth_top design. Problem comes with OpenLane Flow.
Please follow below steps to run my design.

step 1: Hope you have installed openlane flow: https://github.com/The-OpenROAD-Project/OpenLane#installation-notes

step 2: change your current directory into openlane and start mount openlane : $ make mount

step 3: Create new design setup for eth_top design: **bash$** ./flow.tcl -design eth_top -init_design_config
Now check the path ./designs/eth_top/config.tcl and src directory created.
Edit eth_top/config.tcl and  change CLOCK_PORT name "clk" to "wb_clk_i"
Untar https://github.com/vijayank88/vijayank88/blob/main/eth_top.tar.xz this file and src/* netlist file copy under ./designs/eth_top/src/
Now design setup is ready. By default OL takes sky130A PDK library. Type the below command to run the design in OpenLane.

step 4: **bash$** ./flow.tcl -design eth_top
Check the status its completing or stops with error message as I posted.

Thanks...

donn, 2021-08-05:
Packaged. Ball's in your field, OR team.

vijayank88, 2021-08-06:
@donn Yosys default core size 940-955 taken with clock period 10ns. I've manually increased core size to 1200 -1250 with clock of 15ns able to pass routing stage. Testing with 10ns clock period again. Thanks...

vijayank88, 2021-08-06:
@donn i am able to complete eth_top design routing with core area 960-980 with 10ns clock period. Also made repair_design post placement. Able to generate GDSII now without any stop.

Default core taken from OLFS is 940-955.

dralabeing, 2021-08-09:
> Packaged. Ball's in your field, OR team.

Thanks @Donn- @maliberty - Could you please take a look?

donn, 2021-08-12:
OR team confirms issue is fixed in HEAD: would require merging an OpenROAD update to fix this. Will investigate.

dralabeing, 2021-08-12:
@vijayank88 Can you rerun this with ORFS to verify before the merge?

vijayank88, 2021-08-13:
@dralabeing At the time of GH issue filing I checked with ORFS with sky130A_hd PDK RTL to GDSII generation working fine. Issue comes from OLFS with sky130A PDK.

dralabeing, 2021-08-31:
The original testcase did not work in the OL flow- default core size fails. The ORFS message is more actionable the allows flow completion through modification of config.tcl for die size.
There error messages in OL  need to be clear and actionable and issued earlier in the flow.

donn, 2021-08-31:
#547, please don't reopen stale issues for requests already captured.

closed 2021-08-31 by donn

vijayank88, 2021-08-31:
@donn  today updated OL repo at my end and got below error message for this eth_top design.

[INFO GRT-0003] Macros: 0
[INFO GRT-0004] Blockages: 2566655
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 299
[INFO GRT-0017] Processing 1733061 blockages on layer li1.
[INFO GRT-0017] Processing 336278 blockages on layer met1.
[INFO GRT-0017] Processing 13 blockages on layer met4.
[INFO GRT-0017] Processing 13 blockages on layer met5.
[INFO GRT-0009] Nets to reroute: 32103.
[INFO GRT-0101] Running extra iterations to remove overflow.
[WARNING GRT-0164] Initial grid wrong y1 x1 [132 34], net start [56 60] routelen 46.
[WARNING GRT-0165] End grid wrong y2 x2 [132 34], net start [18 52] routelen 46.
[ERROR GRT-0167] Invalid 2D tree for net _19655_.
Error: or_groute.tcl, 68 GRT-0167
[ERROR]: during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_groute.tcl |& tee >&@stdout /openLANE_flow/designs/eth_top/runs/6thRun/logs/routing/18-fastroute.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

[ERROR]: Please check openroad  log file
[ERROR]: Dumping to /openLANE_flow/designs/eth_top/runs/6thRun/error.log
[INFO]: Calculating Runtime From the Start...
[INFO]: flow failed for eth_top/31-08_16-07 in 0h13m7s
[INFO]: Generating Final Summary Report...
[ERROR]: during executing: "openroad -python /openLANE_flow/report_generation_wrapper.py -d /openLANE_flow/designs/eth_top -dn eth_top -t 6thRun -o /openLANE_flow/designs/eth_top/runs/6thRun/reports/final_summary_report.csv -m /openLANE_flow/designs/eth_top/runs/6thRun/reports/manufacturability_report.rpt -rs /openLANE_flow/designs/eth_top/runs/6thRun/reports/runtime_summary_report.rpt -r /openLANE_flow/designs/eth_top/runs/6thRun"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
EOFError: EOF when reading a line
grep: /openLANE_flow/designs/eth_top/runs/6thRun/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/eth_top/runs/6thRun/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/eth_top/runs/6thRun/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/eth_top/runs/6thRun/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/eth_top/runs/6thRun/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/eth_top/runs/6thRun/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/eth_top/runs/6thRun/logs/routing/fastroute.log: No such file or directory

[ERROR]: Please check openroad  log file
[ERROR]: Dumping to /openLANE_flow/designs/eth_top/runs/6thRun/error.log