The-OpenROAD-Project/OpenLane #499: Add -u 0 option instead of  -u $(shell id -u $(USER)):$(shell id -g $(USER)) to enable rootless docker

atorkmabrains, 2021-07-26:
If you use -u 0 instead of -u $(shell id -u $(USER)):$(shell id -g $(USER)) it should enable docker rootless setup allowing more secure environment.

More information about rootless docker:
https://docs.docker.com/engine/security/rootless/

donn, 2021-07-26:
Waiting on PR #500

donn, 2021-07-27:
We'd have to rethink the security model of running OpenLane to achieve this. You saw that even a minute permission change broke the CI. We'd also have to test this change with various setups and see what sticks.

This also causes problems with [podman](https://podman.io/) which just straight up does not handle the `-u` permissions the same way Docker does. So I've run into issues with this personally, and I've found that a decent workaround is simply commenting out the relevant line in the Makefile or running the relevant Docker commands manually.

I'm open to discussion, but this is not a cut and dry change as it looked like it was.