The-OpenROAD-Project/OpenLane #501: set_propagate_clk [all_clocks] not working

vijayank88, 2021-07-26:
**Describe the bug**
in OpenLane flow scripts/openroad/or_resizer_timing.tcl  below lines available. 
=========================================================
# CTS and detailed placement move instances, so update parastic estimates.
estimate_parasitics -placement
set_propagated_clock [all_clocks]
repair_timing
==========================================================

The respective log is 15-fastroute.log don't have propagated delay, still running with ideal delay. Log shown below:

===================================================
Startpoint: _898_ (rising edge-triggered flip-flop clocked by clk)
Endpoint: _884_ (rising edge-triggered flip-flop clocked by clk)
Path Group: clk
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                  0.00    0.00    0.00   clock clk (rise edge)
                          0.00    0.00   **clock network delay (ideal)**
                  0.00    0.00    0.00 ^ _898_/CLK (sky130_fd_sc_hd__dfxtp_1)
                  0.06    0.84    0.84 v _898_/Q (sky130_fd_sc_hd__dfxtp_1)
     2    0.00                           uart_rx_inst.prescale_reg[1] (net)
                  0.06    0.00    0.84 v _485_/A (sky130_fd_sc_hd__or2_1)
                  0.18    0.59    1.43 v _485_/X (sky130_fd_sc_hd__or2_1)
     4    0.01                           _328_ (net)
                  0.18    0.00    1.43 v _486_/C (sky130_fd_sc_hd__or3_1)
                  0.22    0.87    2.31 v _486_/X (sky130_fd_sc_hd__or3_1)
     4    0.01                           _329_ (net)
                  0.22    0.00    2.31 v _487_/D (sky130_fd_sc_hd__or4_1)
                  0.26    1.12    3.43 v _487_/X (sky130_fd_sc_hd__or4_1)
     3    0.01                           _330_ (net)

====================================================

**To Reproduce**

Take any available design in OpenLane flow and check design/<name>/runs/logs/routing/fastroute.log

**Expected behavior**
Once we add set_propagated_clock [all_clocks] this clock dealy should be used in all next stages. If we included in cts, routing stage should be used same propagated clocks. Example log shown below:
=======================================================
Startpoint: _897_ (rising edge-triggered flip-flop clocked by clk)
Endpoint: _874_ (rising edge-triggered flip-flop clocked by clk)
Path Group: clk
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                          0.00    0.00   clock clk (rise edge)
                          1.26    1.26   **clock network delay (propagated)**
                  0.67    0.00    1.26 ^ _897_/CLK (sky130_fd_sc_hd__dfxtp_1)
                  0.10    0.87    2.13 v _897_/Q (sky130_fd_sc_hd__dfxtp_1)
     3    0.01                           uart_rx_inst.prescale_reg[0] (net)
                  0.10    0.00    2.13 v _485_/B (sky130_fd_sc_hd__or2_1)
                  0.15    0.53    2.66 v _485_/X (sky130_fd_sc_hd__or2_1)
     4    0.01                           _328_ (net)
                  0.15    0.00    2.66 v _486_/C (sky130_fd_sc_hd__or3_1)
                  0.20    0.84    3.50 v _486_/X (sky130_fd_sc_hd__or3_1)
     4    0.01                           _329_ (net)
==============================================================

As per standard EDA flow clock propagation delay should be initialized before start of CTS.This clock tree added .def and .lef should be used in routing stage and our tns and wns should be optimized to meet our constraints.

maliberty, 2021-07-26:
This report is mistaken, or_resizer_timing.tcl does not generate 15-fastroute.log, it generates 12-resizer-timing.log (which has no timing report). 

or_groute.tcl does generate 15-fastroute.log but it has no call to set_propagated_clock.  The tool results are correct and this is a flow issue so return to @donn

vijayank88, 2021-07-27:
@maliberty After CTS, or_resizer_timing.tcl running and respective log is 12_resizer_timing.log. I think detailed timing checks not added at this stage. In routing stage again timing checks happening at 15-fastroute.log. In this fastroute.log i see data path checks only ideal clock not a propagated clock.
My question is as we initialized propagation dealy in run_resizer_timing, in routing stage why its been reset to ideal clock? 
Maybe instead of reading base.sdc, run_resizer_timing stage generated ***.sdc if we use this issue will be resolved (Its just my view)

maliberty, 2021-07-27:
The tool is exited and restarted between steps so there is no carry over.

donn, 2021-07-27:
@maliberty Thanks for the tip, Matt- I'll look into it

dralabeing, 2021-08-25:
@donn Please point to the source tcl code to write sdc files at each of the required stages for correct carryover into global route. We are trying to fix it and test locally first. Thanks

commit b4f32f87 (2021-08-30): Adding Clock Propagation (#562)

* Update or_resizer_timing.tcl

Using CTS stage generated sdc in resizer timing optimization. Writing sdc file again to use it in routing stage timing check

* Update or_cts.tcl

Write sdc file after CTS to use this during resizer_timing_optimization

* Update or_groute.tcl

Using post cts generated sdc with propagation clock for timing check at routing stage

Fixes  #501
PR #606 (2021-09-17): Read CTS Generated SDC File When Appropriate
During post routing timing check use post cts generated sdc file with propagated clock. line 27 updated and line 40 commented.

```
#read_sdc -echo $::env(BASE_SDC_FILE)
if {$::env(CLOCK_TREE_SYNTH)} {
	read_sdc -echo $::env(cts_result_file_tag)_1.sdc
} else {
puts "INFO:Skipped CTS Stage so reading base SDC file"
	read_sdc -echo $::env(BASE_SDC_FILE)
}
```

---

Fixes #501.
closed 2021-08-30 by donn