The-OpenROAD-Project/OpenLane #507: Openlane SKY130 error with iverilog for gate level simulation

Preetbatavia, 2021-07-27:
Error on invoking iverilog along with the gate level netlist created by openlane design flow and a verilog file including the sky130 std_cell_library.

Steps to reproduce the behavior:
1. Go to the src folder where you have copied the gatelevel netlist and created the testbench verllog file 
2. In the terminal type : `iverilog -o my_design1  mux_open.v test1_synthesis_preroute.v gate_def.v `

3. 
4. See error : ```
/home/preet/OpenLane/pdks/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v:72934: syntax error
/home/preet/OpenLane/pdks/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v:72934: error: invalid module item.
```
` wire 1 `  is the line 72934 of the file /sky130_fd_sc_hd.v which is the error 

**Expected behavior**
According to the testbench and iverilog a netlist file in .vcd format should be created which will then be ran with gtkwave.

**Desktop (please complete the following information):**
 - OS: Ubuntu Linux 18.04 LTS
 - Openlane Version :v0.15
 - open_pdks Version: [commit hash]
 - skywater-pdk Version: [commit hash]

donn, 2021-07-27:
I think this is a known issue- @Manarabdelaty has a workaround, I believe

Manarabdelaty, 2021-07-27:
@Preetbatavia The issue is coming from the original PDK files. So, the current solution is to manually comment out the `wire 1` statement until this issue https://github.com/google/skywater-pdk/issues/297 is addressed in the skywater-pdk repo.

donn, 2021-07-27:
Well, there you have it.

donn, 2022-09-06:
Confirmed fixed. Not sure when, but it's fixed as of HEAD.

closed 2022-09-06 by donn