The-OpenROAD-Project/OpenLane #509: ROUTING_CORES not working

maliberty, 2021-07-28:
In my design/.../config.tcl I set:
set ::env(ROUTING_CORES) 40

However when the designs runs in TritonRoute I don't see 40 threads working.

**To Reproduce**
Run any design with this modification.

I used OL 3c0f2fd0e5daf1ff1047fa802b7cc53bb8d4f7a7 as my build.

donn, 2021-07-28:
Workaround: Modify `configuration/routing.tcl`

vijayank88, 2021-08-31:
@donn 
Is this routing cores should be less than the available cores in the machine or we can utilize maximum available cores?
For example: in my Centos machine have 4 cores and Ubuntu has 2 cores only.
In Openlane configuration/routing.tcl set default value of 4. So this will affect my ubuntu machine background process right?
So my suggestion we use Makefile $(NPROC) value as threads or exclusively we can call tcl proc in routing.tcl to get number core values. 
In OpenRoad flow NUM_CORES exported from Makefile for set_thread_count $::env(NUM_CORES). Because user donot know about machine configuration and where to use those variables. Plz let me know best usage for OL flow.

donn, 2021-08-31:
Note: This issue will probably be addressed by removing the param file which has been since deprecated.

> Is this routing cores should be less than the available cores in the machine or we can utilize maximum available cores?

Max

vijayank88, 2021-08-31:
@donn thanks

PR #588 (2021-09-07): Rework Multithreading Variables
~ ROUTING_CORES now overridable by environment (set_if_unset)
- Deprecated TR parameter file removed

Fixes #509
closed 2021-09-07 by donn