The-OpenROAD-Project/OpenLane #513: html rendering in regression_results/readme

marwaneltoukhy, 2021-07-29:
**Describe the bug**
Test results should be rendered as html

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'regression_results/'
2. Click on 'sky130_fd_sc_hd'

marwaneltoukhy, 2021-08-05:
`https://htmlpreview.github.io/?` no longer renders csv files in HTML

donn, 2021-08-05:
No, I removed that.

I didn't want for there to be data duplication on our gold standard. People can use Excel, I'm sure.

donn, 2021-08-05:
No, I removed the HTML files in the interest of data de-duplication. htmlpreview does not render CSV files, it merely allows the html files to be rendered instead of being shown as plaintext.

Clearly, I neglected to update the Readme, so thank you for pointing that out. We can probably also utilize some web app that renders the CSV files browser-side.

marwaneltoukhy, 2021-08-05:
github renders CSV files neatly, maybe we could just redirect to the respective CSV file on github.

donn, 2021-08-15:
Possible. Would you consider opening a PR?

marwaneltoukhy, 2021-08-15:
It is added to PR #531 along with timing optimization before global routing

closed 2021-08-31 by marwaneltoukhy