The-OpenROAD-Project/OpenLane #518: Post Synthesis Simulation shows XXX

deepsita, 2021-07-31:
The simulation of synthesized netlist is always XXX. as shown in the image. 
[image]

However, the Synthesis is shown successful as shown in the below image of terminal
[image]

The procedure to reproduce the same is given in the https://github.com/deepsita/mythcore_synth
git clone https://github.com/deepsita/mythcore_synth.git 
cd mythcore_synth/post synth Simulation/
 iverilog gls.v mythcore.synth.v primitives.v sky130_fd_sc_hd.v
 ./a.out
  gtkwave gls.vcd

Manarabdelaty, 2021-08-01:
@deepsita I believe the behavioral models are buggy in the original PDK (see https://github.com/google/skywater-pdk/issues/310) . However, you can simulate with the functional models by passing the FUNCTIONAL define to iverilog (you also need to set UNIT_DELAY macro to some value) :
 `iverilog -DFUNCTIONAL -DUNIT_DELAY=#1 gls.v mythcore.synth.v primitives.v sky130_fd_sc_hd.v`

deepsita, 2021-08-02:
I tried doing the same, and the following errors are shown. 

[image]
[image]

Manarabdelaty, 2021-08-06:
@deepsita See issues (https://github.com/google/skywater-pdk/issues/298, https://github.com/google/skywater-pdk/issues/297), The current solution (until they are addressed in the skywater-pdk repo) is to manually comment out the `wire 1;` statement and correct `endif SKY130_FD_SC_HD__LPFLOW_BLEEDER_FUNCTIONAL_V` to `endif // SKY130_FD_SC_HD__LPFLOW_BLEEDER_FUNCTIONAL_V`

donn, 2022-09-06:
For posterity's sake: the `wire 1` issue has been long fixed, but the `endif most definitely has not.

Asma-Mohsin, 2022-09-09:
i think you should first write powered verilog of this synthesis file cause currently your cells have no power and ground connections. thus going on unknown state.

Jboyrox, 2024-04-30:
Hi I am also facing the same issue . Has there been any updates to this

donn, 2025-10-04:
Everything should be fixed by now. To simulate successfully, make sure that:

- `FUNCTIONAL` is set to `True`.
- If you're simulating a powered netlist (i.e. anything after Synthesis in OpenLane), you need to connect 1 and 0 to VDD/VSS and define `USE_POWER_PINS`.