The-OpenROAD-Project/OpenLane #527: Post synthesis iverilog simulation gives error

manjunathrv, 2021-08-06:
**Describe the bug**
iVerilog simulation for the same test bench gives an error for post-synthesis gate-level netlist. 
The synthesis in Openlane is successful. 
Presynthesis test bench simulation is successful. 

**Expected behavior** </br>
Below Pre-synthesis iverliog output </br>
[image]

**Observed errors**</br>
iverilog simulation for Post synthesis 
[image]

Also tried with adding unit delay commands in iverilog from issue https://github.com/The-OpenROAD-Project/OpenLane/issues/518
[image]

run_synthesis was successful 
[image]

**Desktop (please complete the following information):**
 - OS: [ubuntu]
 - Openlane Version [ 2021.07.31_04.42.28] obtained from get_tag.py
 - open_pdks Version: [latest installed 31st July]
 - skywater-pdk Version: [latest installed 31st July]

armleo, 2021-08-06:
can you provide 5235 +- 10 lines of fs_sc_hd,v?
If it's defined as `wire 1;` then comment it out.

manjunathrv, 2021-08-06:
Thanks for the reply. Yes, it was wire 1; I have commented it. </br>
[image]

iverilog simulation doesn't give errors. But the output as checked in gtkwave shows XXX. </br>

[image]

donn, 2021-08-06:
Duplicate of #518

closed 2021-08-06 by donn