The-OpenROAD-Project/OpenLane #529: Updating Yosys in OpenLane

shariethernet, 2021-08-07:
Hi, How to make openlane use a new version of yosys (0.9+4245). Currently it uses a older version where yosys lacks support for `tri` keyword.
How to edit the dockerfile in  `openlane/docker_build/docker/yosys/Dockerfile` to make openlane use this yosys version
TIA!

d-m-bailey, 2021-08-07:
@shariethernet You should be able to do this from your openlane installation.
1. edit `IMAGE_NAME` in `Makefile` (optional: default is to use the current image)
2. `cd docker_build`
2. edit `IMAGE_NAME` in `Makefile` (optional: default is to use the current image).
3. edit `YOSYS_COMMIT` in `docker/yosys/Dockerfile` to the version you want
4. `make build-yosys`
5. `make merge`

That's it.

shariethernet, 2021-08-07:
I edited `YOSYS_COMMIT` and then ran `make build-yosys`. It was successsful. I am getting this error after running `make merge`

d-m-bailey, 2021-08-09:
@shariethernet not sure, but maybe try `sudo make merge`? Looks like you don't have write access to a file system.

shariethernet, 2021-08-10:
Thanks @d-m-bailey. I just did a `git reset --hard` to undo the changes and repeated the same as `su` worked!

closed 2021-08-10 by shariethernet