The-OpenROAD-Project/OpenLane #536: CentOS7 docker image used in OpenLane docker is deprecated. Git is too old.

armleo, 2021-08-14:
In: https://hub.docker.com/_/centos?tab=tags&page=1&ordering=last_updated you can see that many months the CentOS has not been updated.

Also, the git that is installed in the Docker OpenLane image is outdated. This is while not critical, limits the ability to use OpenLane image to be used to build PDKs, because git pull fails.

d-m-bailey, 2021-08-14:
@armleo There are other issues with using git inside docker (at least on virtual machines). See #532 (I've submitted a pull request with a possible solution).

donn, 2021-08-14:
I am unable to verify your claim that the image has been deprecated. As far as I'm aware, CentOS 7 will continue to be supported through May 2024, and the tools we're using will continue to support it until its EOL.

The git issue I'm aware of, and I did add a manual update to a later git version:

https://github.com/The-OpenROAD-Project/OpenLane/blob/71bf5bd70ec9c2ba20d6d19b01b45f5e2d5c41fe/docker_build/Dockerfile#L80

If you are saying that right now, something is broken, please provide version info/git commit info so we can investigate further, but changing the Docker base image is currently not in the cards.

armleo, 2021-08-14:
Apparently it is not deprecated. Hmm, for some reason I saw centos docker official image in deprecated docker images list and didnt bother to check.
I assumed that's why git was very old.
Both the problems have been addressed. Closing the issue, thanks for your time.

closed 2021-08-14 by armleo