The-OpenROAD-Project/OpenLane #542: [ERROR]: Failure in global placement

vijayank88, 2021-08-16:
**Describe the bug**
I am running riscv_top design with interactive mode. cells count 7L. Its got stop with error 
======
[INFO ODB-0130]     Created 405 pins.
[INFO ODB-0131]     Created 1311396 components and 7962099 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 4448544 connections.
[INFO ODB-0133]     Created 909127 nets and 3513446 connections.
[INFO ODB-0134] Finished DEF file: /openLANE_flow/designs/riscv_top/runs/16-08_13-34/tmp/floorplan/12-pdn.def
[INFO GPL-0002] DBU: 1000
[INFO GPL-0003] SiteSize: 460 2720
[INFO GPL-0004] CoreAreaLxLy: 5520 10880
[INFO GPL-0005] CoreAreaUxUy: 5288620 5293120
[INFO GPL-0006] NumInstances: 1311396
[INFO GPL-0007] NumPlaceInstances: 908992
[INFO GPL-0008] NumFixedInstances: 402404
[INFO GPL-0009] NumDummyInstances: 0
[INFO GPL-0010] NumNets: 909127
[INFO GPL-0011] NumPins: 3513849
[INFO GPL-0012] DieAreaLxLy: 0 0
[INFO GPL-0013] DieAreaUxUy: 5294490 5305210
[INFO GPL-0014] CoreAreaLxLy: 5520 10880
[INFO GPL-0015] CoreAreaUxUy: 5288620 5293120
[INFO GPL-0016] CoreArea: 27906602144000
[INFO GPL-0017] NonPlaceInstsArea: 513207206400
[INFO GPL-0018] PlaceInstsArea: 12561671388800
[INFO GPL-0019] Util(%): 45.86
[INFO GPL-0020] StdInstsArea: 12561671388800
[INFO GPL-0021] MacroInstsArea: 0
Error: out of memory.[ERROR]: Failure in global placement

% df -h
Filesystem               Size  Used Avail Use% Mounted on
overlay                   50G   31G   20G  61% /
tmpfs                     64M     0   64M   0% /dev
tmpfs                    3.9G     0  3.9G   0% /sys/fs/cgroup
shm                       64M     0   64M   0% /dev/shm
/dev/mapper/centos-home  418G  105G  314G  26% /openLANE_flow
/dev/mapper/centos-root   50G   31G   20G  61% /tmp/.X11-unix
tmpfs                    3.9G     0  3.9G   0% /proc/asound
tmpfs                    3.9G     0  3.9G   0% /proc/acpi
tmpfs                    3.9G     0  3.9G   0% /proc/scsi
tmpfs                    3.9G     0  3.9G   0% /sys/firmware
% 
==============

Is there any specific memory required to run such big designs in interactive mode. Given disk space message as well.

**Desktop (please complete the following information):**
 - OS: [e.g. Centos 7]
 - Openlane Version [v0.22-3-g3c0f2fd]

vijayank88, 2021-08-16:
Refer below links for src files and steps to reproduce the issue:
https://app.slack.com/client/T01699QAZBQ/threads

donn, 2021-08-17:
> Refer below links for src files and steps to reproduce the issue:
> https://app.slack.com/client/T01699QAZBQ/threads

Please upload and share the files here.

vijayank88, 2021-08-17:
> > Refer below links for src files and steps to reproduce the issue:
> > https://app.slack.com/client/T01699QAZBQ/threads
> 
> Please upload and share the files here.

https://slack-files.com/T01699QAZBQ-F02BB9S6YG3-2d3ca8daaa

vijayank88, 2021-08-23:
@donn Will try interactive issue later. If I use OpenLane flow scripts its stop at or_replace.tcl. Floorplanned DIE_AREA is "0.0 0.0 5636.69 5647.41". 
Error.log as below:
during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_replace.tcl |& tee >&@stdout /openLANE_flow/designs/riscv_top/runs/1strun/logs/placement/7-replace.log"
Last 10 lines:
child killed: kill signal

during executing: "openroad -python /openLANE_flow/report_generation_wrapper.py -d /openLANE_flow/designs/riscv_top -dn riscv_top -t 1strun -o /openLANE_flow/designs/riscv_top/runs/1strun/reports/final_summary_report.csv -m /openLANE_flow/designs/riscv_top/runs/1strun/reports/manufacturability_report.rpt -rs /openLANE_flow/designs/riscv_top/runs/1strun/reports/runtime_summary_report.rpt -r /openLANE_flow/designs/riscv_top/runs/1strun"
Last 10 lines:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
EOFError: EOF when reading a line

donn, 2021-08-31:
If you want to address the fanout problem, please introduce a new issue. 

As for the memory thing, you can set the memory limit by setting DOCKER_MEMORY=8g or 16g before running `make mount`.

donn, 2021-09-01:
User reported via private channel that memory option is not working.

Will investigate.

tspyrou, 2021-09-01:
@vijayank88 please try the dockerless local install suggestion and close if this works otherwise let us know and re-assign to Donn.

tspyrou, 2021-09-01:
@dralabeing fyi

vijayank88, 2021-09-02:
@tspyrou @donn  thanks for the update. I will try at my and comeback to you.

vijayank88, 2021-09-02:
> User reported via private channel that memory option is not working.
> 
> Will investigate.

@donn With docker mode before mount able to set DOCKER_MEMORY

donn, 2021-09-02:
> its reflecting as well

Please elaborate

vijayank88, 2021-09-03:
@donn My desktop has 16gb RAM. But by default it taken 9gb only. After using
$export DOCKER_MEMORY=16gb
$make mount
while mounting --memory=16gb added and peak memory increased to 13GB

donn, 2021-09-03:
…so did it work or not?

vijayank88, 2021-09-04:
@donn  with increase in memory also not helping. Stopped at placement stage again.
[NesterovSolve] Iter: 440 overflow: 0.818874 HPWL: 40368954113
[NesterovSolve] Iter: 450 overflow: 0.809816 HPWL: 41150573794
[NesterovSolve] Iter: 460 overflow: 0.784982 HPWL: 44127264261

during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_replace.tcl |& tee >&@stdout /openLANE_flow/designs/riscv_top/runs/Sep2Run/logs/placement/7-replace.log"
Last 10 lines:
child killed: kill signal

vijayank88, 2021-09-08:
@donn with latest build had improvement in log. its passed nesteroSolve and stuck with below error. Any idea to fix this?

```
[INFO GRT-0053] Routing resources analysis:
          Routing      Original      Derated      Resource
Layer     Direction    Resources     Resources    Reduction (%)
---------------------------------------------------------------
li1        Vertical     11497500      10654189          7.33%
met1       Horizontal   15330000      12057593          21.35%
met2       Vertical     11497500      11487000          0.09%
met3       Horizontal    7665000       7661484          0.05%
met4       Vertical      4599000       4530826          1.48%
met5       Horizontal    1533000       1530374          0.17%
---------------------------------------------------------------

[ERROR]: during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_replace.tcl |& tee >&@stdout /openLANE_flow/designs/riscv_top/runs/IntRunSep8/logs/placement/7-replace.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: kill signal

[ERROR]: Please check openroad  log file
[ERROR]: Dumping to /openLANE_flow/designs/riscv_top/runs/IntRunSep8/error.log
[INFO]: Calculating Runtime From the Start...
[INFO]: flow failed for riscv_top/08-09_09-40 in 3h9m34s
[INFO]: Generating Final Summary Report...
[ERROR]: during executing: "openroad -python /openLANE_flow/generate_reports.py -d /openLANE_flow/designs/riscv_top -dn riscv_top -t IntRunSep8 -o /openLANE_flow/designs/riscv_top/runs/IntRunSep8/reports/final_summary_report.csv -m /openLANE_flow/designs/riscv_top/runs/IntRunSep8/reports/manufacturability_report.rpt -rs /openLANE_flow/designs/riscv_top/runs/IntRunSep8/reports/runtime_summary_report.rpt -r /openLANE_flow/designs/riscv_top/runs/IntRunSep8"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
sed: can't read /openLANE_flow/designs/riscv_top/runs/IntRunSep8/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/riscv_top/runs/IntRunSep8/logs/routing/fastroute.log: No such file or directory
Traceback (most recent call last):
  File "/openLANE_flow/generate_reports.py", line 61, in <module>
    report = Report(design, tag, design_name, params, run_path).get_report()
  File "/openLANE_flow/scripts/report/report.py", line 133, in get_report
    self.formatted_report = self.format_report()
  File "/openLANE_flow/scripts/report/report.py", line 120, in format_report
    splited_report[dieareaIdx] = str(float(splited_report[dieareaIdx])/1000000)
ValueError: could not convert string to float: 'E404'

[ERROR]: Please check openroad  log file
[ERROR]: Dumping to /openLANE_flow/designs/riscv_top/runs/IntRunSep8/error.log

```

maliberty, 2021-09-13:
@eder-matheus made some memory usage improvements in grt - I'm not sure if OL has a current version of OR with those improvements or not.  What git commit are you using in this testing?

vijayank88, 2021-09-14:
@maliberty  git version 2.25.1
OL tag: 2021.09.03_03.49.05

maliberty, 2021-09-14:
@donn how can I convert an OL tag to an OR commit?

donn, 2021-09-14:
`https://github.com/The-OpenROAD-Project/OpenLane/blob/<your OL tag here>/dependencies/tool_metadata.yml`

You’ll find the commit for OpenROAD under the entry `openroad_app`.

maliberty, 2021-09-14:
I see f0e6c04818f120fcd15c29735a398d3cd420bd78 for OR which is from July 28.  @eder-matheus were your improvements after that?

eder-matheus, 2021-09-14:
> I see f0e6c04818f120fcd15c29735a398d3cd420bd78 for OR which is from July 28. @eder-matheus were your improvements after that?

My PR with the fixes for memory leaks was merged in Aug 2.

maliberty, 2021-09-14:
@donn this is back to you for an OR upgrade.

maliberty, 2021-09-15:
Actually 2021.09.15_03.22.49 seems to have a later version of OR.  @vijayank88 can you try that version?

vijayank88, 2021-09-15:
still stops at same error:
[NesterovSolve] Iter: 450 overflow: 0.820241 HPWL: 40104038173
[NesterovSolve] Iter: 460 overflow: 0.801261 HPWL: 43521142110
[NesterovSolve] Iter: 470 overflow: 0.789143 HPWL: 49929210183
[ERROR]: during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_replace.tcl |& tee >&@stdout /openLANE_flow/designs/riscv_top/runs/Sep15Run/logs/placement/7-replace.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: kill signal
.....
invoked from within
"try_catch $::env(OPENROAD_BIN) -exit $::env(SCRIPTS_DIR)/openroad/or_replace.tcl |& tee $::env(TERMINAL_OUTPUT) [index_file $::env(replaceio_log_file_..."
    (procedure "global_placement_or" line 14)
    invoked from within
"global_placement_or"
    (procedure "run_placement" line 17)
    invoked from within
"run_placement"
    (procedure "run_placement_step" line 11)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 43)

maliberty, 2021-09-15:
@vijayank88 previous you said "its passed nesteroSolve" so this is different from above.  Does the previous solution to that error not work anymore?

vijayank88, 2021-09-15:
@maliberty if we use default size calculated from synthesis stage got failing with nestrosolve.
If I re-floorplan with manual size its get stop with below message:

[INFO GRT-0053] Routing resources analysis:
          Routing      Original      Derated      Resource
Layer     Direction    Resources     Resources    Reduction (%)
---------------------------------------------------------------
li1        Vertical     11497500      10654189          7.33%
met1       Horizontal   15330000      12057593          21.35%
met2       Vertical     11497500      11487000          0.09%
met3       Horizontal    7665000       7661484          0.05%
met4       Vertical      4599000       4530826          1.48%
met5       Horizontal    1533000       1530374          0.17%
---------------------------------------------------------------

[ERROR]: during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_replace.tcl |& tee >&@stdout /openLANE_flow/designs/riscv_top/runs/IntRunSep8/logs/placement/7-replace.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: kill signal

Consider still flow not able to complete global_placement_or stage.

vijayank88, 2021-10-07:
@dralabeing this design has 3 top level module. I hardened 1 core module and other 2 I used as it is. Still its failing at placement stage. I will blackbox all top level module and try it. I will update once flow complete successfully. Thanks

donn, 2021-12-12:
Staled out.

closed 2021-12-12 by donn