The-OpenROAD-Project/OpenLane #543: or_issue.py doesn't capture the full environment needed to run

maliberty, 2021-08-16:
**Describe the bug**
It seems that or_issue.py doesn't capture the full environment needed to run.  In particular I'm looking at a case using or_replace.tcl which depends on lots of envars but the run script doesn't capture them.  The first example is MERGED_LEF_UNPADDED.

Would you fix the script to capture all required envars?

**To Reproduce**
Steps to reproduce the behavior:
1. run or_issues.py on any testcase and capture it at the or_replace.tcl stage

**Expected behavior**
A reproduceable test case

donn, 2021-08-17:
Aye-aye: as I've mentioned, this is a very best-effort script. I'm not surprised that there are blind spots.

donn, 2021-08-17:
Can't replicate. What command are you running, exactly?

maliberty, 2021-08-17:
I ran 

python3 or_issue.py /openLANE_flow/designs/riscv_top -s scripts/openroad/or_replace.tcl -i /openLANE_flow/designs/riscv_top/runs/16-08_16-42/tmp/floorplan/6-pdn.def

using a design directly provided by a user.  I don't think the problem is design specific though.  What did you run that produced a run script with MERGED_LEF_UNPADDED in it?

donn, 2021-08-18:
You passed the path to the design, not the run.

`python3 or_issue.py /openLANE_flow/designs/riscv_top` -> `python3 or_issue.py ./designs/riscv_top/runs/16-08_16-42/`

I will revise the documentation so I am more explicit about what you need to provide.

donn, 2021-09-29:
Should be resolved now.

closed 2021-09-29 by donn