The-OpenROAD-Project/OpenLane #545: Getting the error as or_ioplacer.tcl when 4-ioPlacer.log excution in Openlane

ksooryakrishna1, 2021-08-18:
I am getting the below error when i run my design. How to solve this error?
Error: or_ioplacer.tcl, 34 invalid command name "ppl::set_hor_length"
[ERROR]: during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_ioplacer.tcl |& tee >&@stdout /openLANE_flow/designs/picorv32a/runs/first_run/logs/floorplan/4-ioPlacer.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

[ERROR]: Please check openroad  log file
[ERROR]: Dumping to /openLANE_flow/designs/picorv32a/runs/first_run/error.log

donn, 2021-08-18:
`git remote get-url origin` and `git show HEAD` please.

ksooryakrishna1, 2021-08-18:
> `git remote get-url origin` and `git show HEAD` please.

not understood what i have to do

donn, 2021-08-18:
Please run these two commands and post the output.

ksooryakrishna1, 2021-08-19:
For the command - git remote get-url origin, I get -
https://github.com/efabless/openlane.git

For the command - git show HEAD, I get - 
commit 880ade4b4d88de8bb4b91e6b80d663c8294c1082 (HEAD, tag: v0.12)
Author: Travis CI <travis@travis-ci.org>
Date:   Wed Mar 24 04:13:20 2021 +0000

    Update Latest Green Tag to v0.12

diff --git a/Makefile b/Makefile
index e28b1fb..b8cb534 100644
--- a/Makefile
+++ b/Makefile
@@ -41,7 +41,7 @@ SPECIAL_VOLTAGE_LIBRARY ?= sky130_fd_sc_hvl
 IO_LIBRARY ?= sky130_fd_io
 INSTALL_SRAM ?= disabled
 
-IMAGE_NAME ?= efabless/openlane:v0.11
+IMAGE_NAME ?= efabless/openlane:v0.12
 TEST_DESIGN ?= spm
 BENCHMARK ?= regression_results/benchmark_results/SW_HD.csv
 REGRESSION_TAG ?= TEST_SW_HD
diff --git a/README.md b/README.md
index 9a20525..2a5a14b 100644
--- a/README.md
+++ b/README.md
:

donn, 2021-08-19:
You're using the wrong link. Please clone from https://github.com/The-OpenROAD-Project/OpenLane instead and try again.

Cheers.

closed 2021-08-19 by donn