The-OpenROAD-Project/OpenLane #547: Do improvement in Error info if project fails at any stage

vijayank88, 2021-08-18:
**Is your feature request related to a problem? Understanding of log message in OpenLane log
I am running one of sample design in openlane. Its got failed at routing stage. But there is no clear information why its stopped. Same project i ran using OpenRoad flow scripts its failed and gave info as default placement density is .55 and expected PL_DENSITY is 0.6. or modify core area.

**Describe the solution you'd like**
If project fails means display Error number and possible workaround to fix this issue. Like above increase core area or placement density value.

[image] This image captured from OpenRoad flow command log.
[image] This logs captured from OpenLane flow.

Both run failed due to placement density and core area. But error info was clear in OpenRoad scripts, not in OpenLane flow scripts. So if you had this enhancement in OLFS its helpful to user to understand the root cause quickly.

donn, 2021-08-18:
These two appear to be two distinct failures in two different parts of the flow. I don’t see what’s actionable here.

vijayank88, 2021-08-19:
@donn 
I start my design with OpenLane and tool calculating core size on its own based on number of cells. OL calculated  core size is 0 0 940 950. But its failed with below message 
------------
Error 0: order_wires failed
Notice 0: error, net 26404, shorts to another term
Notice 0: wire point (447385 896300)
Notice 0: clknet_leaf_256_wb_clk_i
--------
Refer my previous issue #497 
So from OpenLane I got core area, used this value in OpenRoad flow its stopped with Error message and suggesting to re-floorplan or increase PL density.

Again I re-floorplan with  0 0 960 980 I able generate GDSII. Even OpenLane has able to pass this Error 0: order_wires failed, and GDSII file got generated.

So from above OpenLane Error message order_wires failed no where pointing to re-floorplan our design. But openroad does pointing to use re-floorplan.

So this is what my suggestion to do an enhancement in the Openlane scripts to print such user understandable error info.
As I did comparison with both flow I got this information. If people using only OpenLane will post GitHub issue for everything.

closed 2021-12-29 by donn

MujBilal, 2022-09-28:
LVS reports:
    net count difference = 0
    device count difference = 0
    unmatched nets = 0
    unmatched devices = 0
    unmatched pins = 29
 property failures = 0

Total errors = 29