The-OpenROAD-Project/OpenLane #548: Klayout GDSII PNG

Badboy1307, 2021-08-18:
### Prompt
Bringing Back Klayout generating GDSII image in PNG format will help a lot of students who are using this flow. As I find Klayout only GDS files but not including PNG format. Pls look on this feature.

donn, 2021-08-18:
All you'd have to do is add `set ::env(TAKE_LAYOUT_SCROT) 1` to your design's config.tcl.

This feature has been known to crash, so that's why we've disabled it by default.

Badboy1307, 2021-08-18:
Oh ok thanks for replying will try and let ya know how it's working.

Badboy1307, 2021-08-18:
Is it normal while installing  pdk to run out of range? Or is there a fix for that?

donn, 2021-08-18:
No, not really. You’d have to share the full error (in a new issue) so we can help with that.

Badboy1307, 2021-08-18:
Sure will do so as soon as possible

closed 2021-08-18 by Badboy1307