The-OpenROAD-Project/OpenLane #550: Error is coming in placement while executing my design

vanishka-tanwar, 2021-08-19:
As, I am executing my design in openlane then error is coming in placement and I don't able to resolve this issue 

I have used           ./flow.tcl -design dvsd_cmp        this command for execution but error is coming in placement It is showing this error 
[image]
[image]
[image]

donn, 2021-08-19:
Hello Vanishka, is there any particular reason you did not fill out the issue template?

vijayank88, 2021-08-19:
@vanishka-tanwar  attach your RTL or its small code.. u can copy paste here

vijayank88, 2021-08-19:
@donn  even i tested that code in my latest openlane version. its tiny design with cell count 15. Its failing at detailed_placement_or stage.

donn, 2021-08-19:
Well, thank you for verifying the version, but I do need the design in order to isolate the issue.

vijayank88, 2021-08-19:
@donn  copy the below code. design has no clock.

--------
//***************************************
// Vanshika Tanwar
//Verilog file: dvsd_cmp.v
//***************************************
module dvsd_cmp(
        //Input A        
A_in,                                                                                      
        //Input B
B_in,                                                      
        //When A is less than B, it is high
less_than,                                                                                                                                  
        //When A is equal to B, it is high
equal_to,
        //When A is greater than B, it is high
greater_than                                                                                                                                                                                                                                                      
);
        //Declaration of Input Ports.
    input [3:0] A_in;
    input [3:0] B_in;
        //Declarations of Output Ports.
    output less_than;
    output equal_to;
    output greater_than ;
        //Internal variables
        //reg declarations
     reg less_than;
     reg equal_to;
     reg greater_than;
        //When the inputs  A or B are changed then execute this block
    always @(A_in or B_in)
 begin
        //This block checks if A is Greater than B, Greater than is high
       if(A_in > B_in)                              
           begin                                        
                      less_than = 4'b0;      
                      equal_to =4'b0;
                      greater_than = 4'b1;    
           end
        //This block checks if A is equal to B, Equal to is high
       else if(A_in == B_in)                    
           begin                            
                      less_than = 4'b0;
                      equal_to = 4'b1;
                      greater_than = 4'b0;    
           end
        //Otherwise -This block checks for A less than B, Less than is high          
       else                                                  
           begin                                                  
                      less_than = 4'b1;
                      equal_to = 4'b0;
                      greater_than =4'b0;
        end
end
        //End of a module
endmodule
-------------

vijayank88, 2021-08-19:
@donn  with this openlane 2021.08.17_03.26.57 version able run successfully. I got struck with openlane v0.22-3-g3c0f2fd throwing error in detailed routing. We can ask @vanishka-tanwar  to update to latest version maybe.

vijayank88, 2021-08-19:
@donn 
//***************************************
// Vanshika Tanwar
//Verilog file: dvsd_cmp.v
//***************************************
module dvsd_cmp(
        //Input A        
A_in,                                                                                      
        //Input B
B_in,                                                      
        //When A is less than B, it is high
less_than,                                                                                                                                  
        //When A is equal to B, it is high
equal_to,
        //When A is greater than B, it is high
greater_than                                                                                                                                                                                                                                                      
);
        //Declaration of Input Ports.
    input [3:0] A_in;
    input [3:0] B_in;
        //Declarations of Output Ports.
    output less_than;
    output equal_to;
    output greater_than ;
        //Internal variables
        //reg declarations
     reg less_than;
     reg equal_to;
     reg greater_than;
        //When the inputs  A or B are changed then execute this block
    always @(A_in or B_in)
 begin
        //This block checks if A is Greater than B, Greater than is high
       if(A_in > B_in)                              
           begin                                        
                      less_than = 4'b0;      
                      equal_to =4'b0;
                      greater_than = 4'b1;    
           end
        //This block checks if A is equal to B, Equal to is high
       else if(A_in == B_in)                    
           begin                            
                      less_than = 4'b0;
                      equal_to = 4'b1;
                      greater_than = 4'b0;    
           end
        //Otherwise -This block checks for A less than B, Less than is high          
       else                                                  
           begin                                                  
                      less_than = 4'b1;
                      equal_to = 4'b0;
                      greater_than =4'b0;
        end
end
        //End of a module
endmodule

vijayank88, 2021-08-19:
@donn  with PL_TARGET_DENSITY 0.8 able to resolve the above issue. Default taken 0.6 @vanishka-tanwar  try this update in config.tcl

vanishka-tanwar, 2021-08-20:
Yes, after updating config .tcl , now it is running smoothly in latest version .

I am closing this issue now .

closed 2021-08-20 by vanishka-tanwar